libcsp/libcsp #48: I2C message receive

atrib, 2015-05-20:
I am confused as to how packets should be received over I2C.

The i2c interface defines the following:
/**
 * When a frame is received, cast it to a csp_packet
 * and send it directly to the CSP new packet function.
 * Context: ISR only
 * @param frame
 */
void csp_i2c_rx(i2c_frame_t * frame, void * pxTaskWoken)

Should this be called by the ISR after an entire packet has been received?
Does that mean that we have to manually reconstruct the frame (length, etc) before calling this function?
What is the parameter- pxtaskwoken?

Meanwhile: I remember having read somewhere that to make the I2C driver I only need to define the i2c_init/send/recv functions. What is the prototype for i2c_recv? Its not there in i2c.h.

johandc, 2015-05-20:
Yes.

If you want to receive frames and have an interrupt driven driver the pxtaskwoken must point to a variable otherwise it should be null.

There is no i2c recv function. This will be your own Isr.

You must reconstruct the frame length field and insert the data.

atrib, 2015-05-22:
What is the difference between the fields
  len
  len_rx 
in the struct i2c_frame_t ?
Which one should be set when I am receiving packets?

Finally, I am setting data, len and dest fields in the received frame to be sent to csp_i2c_rx function.
However, when running the test server program from examples/simple.c, I am getting the error
FREE: Invalid CSP buffer pointer 0xc0
from the csp_buffer_free call in task_server function. Do you know why this might happen?

johandc, 2015-05-22:
The I2C frame should be copied into a csp_buffer before the call to new_packet otherwise you will get an error. You can also do it zero-copy by allocating the csp-buffer whithin the I2C driver. If you notice the structure of the i2c_frame and csp_packet they are made so that the i2c frame can later be casted into a csp packet.

The len field must be set by the receiving node.
The len_rx field is used for outgoing master-transmit frames (in i2c_transaction) where a followup with master-receive is required. This field is not used for normal csp operation.

atrib, 2015-05-22:
Thanks.

Finally, how do I add the new drivers to the waf build system?
The i2c driver has its own file in src/drivers/i2c

johandc, 2015-05-25:
The simple method: Open the file called 'wscript' and see if you can add a source file. If you need help, there is some excellent online documentation available for waf. 
The right method: You can also build the driver within your application and then link it with libCSP at a later stage in the build process.

closed 2015-05-25 by johandc