libcsp/libcsp #52: CAN/ driver doesn't receive interface! multiple can interfaces on the same microcontroller not supported.

nicob87, 2015-07-06:
Hi!, I have implemented our drivers for your can interface, and it is working perfectly, but I have found an issue:
    We have 2 can peripherals in our micro-controller, so we need to create 2 can interfaces, and route them independently to different destinations.
    I think this is happening because all of your hardware nodes only have "one" interface of each type. And yo have never deal with this issue.
    To fix this problem some modifications needs to be done:

1) User needs to be able to create several instances of an interface. Right now interface is just a global variable declared in csp_if_can.c. Each interface would have at list a different name "can0", "can1".

2) "can_send" function needs to receive as parameter a pointer to the interface. And also the tx_callback. Right now I see you are receiving the interface in the csp_can_tx function, but you don't use it for nothing!

3) csp_can_init function, maybe should be split into 2 functions, one to be called only once for initializing queues and rxtask, and other to be called once for each interface....

If you are ok,  I can try to do some refactor to implement this modifications, and that way I will be more sure about all modifications involved in fixing this issue. And the I show you the results. I won't be able to test all the drivers! I will only modify the signature of them.

I would like to know what do you think about this. Maybe you are already working on this...

Best Regards! Nicolas.

johandc, 2015-07-06:
Hi Nicolas.

Great to hear that you have success with the can driver.

You are right we never came by a situation where the driver needed to have multiple instances for can, but we have for the kiss interface.

So you might get some inspiration on how that interface is made for multiple instances.

We found an issue with the can driver just recently so make sure you are forking from the latest master version.

We are planning on giving the driver an overhaul after the summer break, so I guess around September we will be giving it a new look.

It would be great if you would make a head start on some of the changes.

The multi instance
Locking inside the driver
Asynchronous versus synchronous send calls
Better handling of errors in the hardware
And finally a version that can run without a task.

Best regards
Johan.

nicob87, 2015-07-06:
Great, will let you know as soon as I have something to show.

jledet, 2015-12-07:
The multiple interface instances would be useful for all interfaces. I have opened a new issue to track the progress here: https://github.com/GomSpace/libcsp/issues/61. I'm closing this issue for now.

closed 2015-12-07 by jledet

nicob87, 2015-12-09:
Hi! I agree the right solution is to do a major refactor to all interfaces, instead of just modifying one. Unfortunately we have no time to try to do that, so for now we are already using my branch, I hope in the future be able to merge with your master again. 
Thanks for all the reviews you have done to my pull request. It has been really helpful.

Regards! Nicolás.