libcsp/libcsp #65: waf configure, program [...] is not executable

magalno, 2016-02-25:
When trying to configure csp with avr32 toolchain and freertos, by following the procedure described in https://github.com/GomSpace/libcsp/blob/master/INSTALL.rst, the waf system produces the following error message: "Program ['avr32-gcc'] is not executable". 

I have earlier successfully configured and build csp for avr32 using older version of waf (v.1.6.9) and wscript (1.0.1), but I'm not able to get the current versions to work.

What do I need to do to build csp with freertos for the avr32 toolchain with the current version of csp?
[config_log.txt](https://github.com/GomSpace/libcsp/files/146569/config_log.txt)

johandc, 2016-02-25:
What happens if you try to run the cross-compiler manually:

avr32-gcc --version

I suspect a problem with your PATH setting or the compiler.

magalno, 2016-02-25:
I too suspected this at first, though it looks like the toolchain is added to path (see the supplied config_log.txt). I am also able to configure and build with older versions of csp/waf/wscript. 

Running the avr32gcc --version command returns "avr32-gcc.exe (AVR_32_bit_GNU_Toolchain_3.4.2_435) 4.4.7" plus a disclaimer.

closed 2016-03-03 by magalno