libcsp/libcsp #70: Proper bindings

jledet, 2016-04-14:
As correctly pointed out by issue #54, we need a better way of generating the bindings. Something with support for multiple languages would be nice, but python bindings should be our main focus.

johandc, 2017-03-26:
@perwulff @sorennohr I have marked this as comming in v1.5. From what i've seen in your pr #92 it lookeds pretty solid already. So i was hoping you could make it?

perwulff, 2017-04-11:
Yes, we are working on it internally and once we are satisfied, it will be pushed.

johandc, 2017-08-18:
Hi Per,

What's the status on that python module?

perwulff, 2017-08-21:
It is not ready to be merged into 1.5. Best guess at the moment, is within a month - maybe sooner.

smorad, 2017-11-10:
Has there been any update with this? Ctypesgen does not work with python3.

sorennohr, 2017-11-29:
See https://github.com/libcsp/libcsp/pull/112

closed 2018-01-24 by johandc