libcsp/libcsp #93: Remove waf configure hell

johandc, 2017-03-27:
**We seriously have too many options by now.**

A lot of them are coming from the fact that i like static allocations. Mmmm... See how i'm loving it :)
However, a problem with them is that they need a pre-defined size,  which leads to all of these weird `CSP_SOMETHING_MAX` defines. We even have dynamic allocation for queues (like the csp_conn_queue) (it's dynamic because of FreeRTOS which btw. now supports static allocation as well) and even in this case, we still have a `CSP_CONN_QUEUE_LENGTH` define to set that up. 

All that being said, our `csp_init()` function takes one argument only (the address). We also have a set of functions, get/set hostname, model, revision and even a get/setter for the address. These functions must always be called, otherwise the `ident` service will crash when hitting a null or garbage string.

So yes, i think there is room for improvement here...

johandc, 2017-03-27:
Okay, so here's my take on a solution.

**Multiple init calls**

Let's take a brief look at what `csp_init` does today:

```c
	/* Initialize CSP */
	csp_set_address(address);

	ret = csp_conn_init();
	if (ret != CSP_ERR_NONE)
		return ret;

	ret = csp_port_init();
	if (ret != CSP_ERR_NONE)
		return ret;

	ret = csp_qfifo_init();
	if (ret != CSP_ERR_NONE)
		return ret;
```

I really believe we should have, these calls available to the user with the different options as either arguments or a datastructure. 

**Here's my proposal:**

```c
/* Memory allocation */
csp_buffer_init(count, size); //! This call is already called by user before csp_init()
csp_conn_init(conn_max, conn_queue_length, default_so_mask);
csp_port_init(bind_port_max);
csp_qfifo_init(input_queue_len);

/* Set ident strings and addr */
csp_set_addr(10);
csp_set_hostname("flicka");
csp_set_model("hest");
csp_set_revision("v1.0");

/* Options */
csp_rdp_set_opt(window_size, conn_timeout_ms, packet_timeout_ms, delayed_acks, ack_timeout, ack_delay_count); //! These should really not be globals.

/* Interface configuration here */
csp_if_lo_init();
csp_if_can_socketcan_init(ifc, 1000000, 0); //! Drivers are init'ed by the user
csp_if_i2c_atmel_something_init(I2C0, 400000);
csp_if_kiss_init();

/* Startup */
csp_route_start_task(stack_size, priority);
```

This will increase the modularity in csp by having the buffers, conns, ports, and router qfifo being initialized independently. 

This will be more work for the user, but i belive that it's easier for the user to setup this, than setting a million compile time options :)

This will get rid of the following options:

>     --with-rdp-max-window=SIZE
>     --with-max-bind-port=PORT
>     --with-max-connections=COUNT
>     --with-conn-queue-length=SIZE
>     --with-router-queue-length=SIZE
>     --with-connection-so=CSP_SO

jledet, 2017-04-10:
Completely agree that we have way to many (compile-time) options.

Instead of having the user call each of the "sub-init" functions, how about having an opaque configuration struct with all the options, that is then passed to `csp_init()`. We could have a `csp_conf_set_defaults(&conf)` function that sets sane defaults in the struct, but allows the user to override them later (probably through get/set functions for each option).

This allows the configuration struct to be extended in the future, and the user will automatically get the default setting for new options.

Also agree that the RDP options should be per-connection.

perwulff, 2017-04-11:
Yes, out with compile-time options and in with runtime configuration, so a single build/release have all the functionality (especially on Linux).
I like the idea about the configuration struct (e.g. ASF, pthread, etc.). If the struct is opaque (forward declared struct), we need to implement more code - but its content is 100% changeable.  A public known struct can be extended - but once members are known, they can't be changed. On the other handl, you don't need getters/setters.
We also discussed using a struct for holding all externally CSP required functions, instead of using weak symbols.

johandc, 2017-04-19:
Hehe, i'm glad you suggested the struct. It was already in my proposed answer, but i cut that part out since it was getting way too long.

So first i'll go ahead and make each module accept arugments like i suggested, then afterwards i'll make a csp_init call which sets up the modules with a csp_conf_t and put it in a file called csp_init.c. This will work as an init helper for simple applications that just need the default things setup. I think it's a great idea to match ASF's csp_conf_get_defaults(&conf) style as well.

Should i include csp_router_start_task in csp_init()?

perwulff, 2017-04-20:
Just to clear about initialization, I think initialization should be modular to the extend it makes sense. So "add on/plug in" modules should have their own initialization method (e.g. struct), which hooks them into the CSP core. If we tie everything together in "one" initialization function, the linker can't throw away unused stuff, when building for a static platforms.
So I'm thinking one csp_init(struct*), which configure the CSP core and then additional csp_module_init(struct*), which configures/enables additional modules/features in CSP. It may be necessary to have a csp_start(), which starts/enables the CSP "engine", if that makes sense.

PR #98 (2018-08-21): Remove compile-time configuration options
Fixes #93 

Configuration struct:
I have created a configuration struct. It's not opaque, we need to discuss this. I don't believe it has to be opaque necessarily. 

The configuration struct is copied on csp_init, and shared to all csp modules through a global csp_conf symbol declared in csp_init.h. We could perhaps eliminate the use of a global if we wish. It would require internal functions in csp_conn_innit() and friends to take more arguments.

Allocations:
Generally i have replaced all static allocations with one-time calls to calloc/malloc.

Note [... cut]
closed 2018-08-21 by johandc