libcsp/libcsp #94: Unused functions and variable - remove?

perwulff, 2017-04-18:
These functions and variable are not used in libcsp. Is there a greater scheme - or simply remove them?

file: libcsp/include/csp/drivers/usart.h: 
int usart_messages_waiting(int handle);
static inline int usart_stdio_msgwaiting(void) {
	return usart_messages_waiting(0);
}

file: libcsp/src/drivers/usart/usart_linux.c
int usart_stdio_id = 0;

johandc, 2017-04-19:
Well it was needed by GOSH in order to do non-blocking reads (for example to break out of a while loop). On some systems there are more than one USART and GOSH could potentially be running on either of them. So i think this is a leftover from that time. Slash uses a waitfunc callback pattern to solve this problem.

If you don't need them any more, you can safely cut that out.

perwulff, 2017-04-20:
What is "slash"?

GOMspace doesn't use it anymore, because it caused everything to rely on libcsp for linux - so we changed that. 
I will remove them in near future then, so we keep libcsp "clean"  :)

johandc, 2017-04-20:
Slash (or satlab shell) was Jeppes first take at building a shell. (from before he worked at gomspace).
https://github.com/satlab/slash
It's open source now.
Jeppe originally built it for linux only. But i adapted it for our use on Space Inventor products and have added tab-completion for commands.

closed 2019-06-20 by perwulff