libcsp/libcsp #95: Propose changing the naming of waf/wscript build task(s)

perwulff, 2017-04-20:
Today the build task named 'csp' can generate 3 different outputs (objects, static library or shared library), depending on the options --enable-bindings and --disable-stlib.
--disable-stlib decides whether to create objects or static library.
--enable-bindings over-writes the former object/static build task (may not be intentional though) and creates a shared library.
This also means, that only one of the above can be built within a given configuration.

I would like to be able to build both objects/static and shared in the same configuration. As far as I understand Waf, this will require that the build tasks have unique names. Proposal:
'csp_h' header files (as today)
'csp' objects/static library (as today) -> libcsp.a
'csp_shlib' shared library -> libcsp.so.

Python bindings will require shared library, so the code must be built with -fPIC). By building both, other targets can either use = 'csp' (for static) or 'csp_shlib' (for shared). Of course the static version can be built with -fPIC (on Linux or if bindings enabled), but thats normally not the how a static library is built. 

What do you guys think?

johandc, 2017-04-21:
I agree, that is a fine solution.

jledet, 2017-04-28:
Yes, agree as well :+1:

cevans01, 2017-05-25:
Is there currently a waf build task to install headers? Something such as the 'csp_h' proposed target above? The following will install the .so in /usr/local/lib but will not install headers

sudo ./waf configure --with-os=posix --prefix=/usr/local --enable-bindings --install-csp build install

perwulff, 2017-05-30:
As it is right now, header files should be installed in <prefix>/include/csp/...., in your case /usr/local/include/csp/...
Or do you mean a command for "only" installing the headers?

cevans01, 2017-05-30:
I believe I may have had conflicting headers from https://github.com/daniestevez/gr-csp 

Seems like the install-csp waf build task is working properly, thanks!

closed 2017-08-21 by perwulff