libcsp/libcsp #96: Possible socket allocator memory leak

bandren, 2017-06-02:
In the function [```csp_socket()```](https://github.com/libcsp/libcsp/blob/17bbef37142d67ebb8bf87044296bf6a79d3a799/src/csp_io.c#L173-L185) of the current commit in master at the time of writing this.

A socket is allocated using the ```csp_conn_allocate( )```. Then, if connectionless, a queue is allocated for the socket using ```csp_queue_create( )```. If the queue failed to allocate then the function returns without freeing the socket allocation.

I'm not sure if this is the intended functionality. If it is, just delete this issue.

perwulff, 2017-06-11:
It seems to be a bug, as the reference to 'sock' is lost (not returned). Will look into that.

perwulff, 2018-08-29:
Fixed in branch release-1.5.

closed 2018-08-29 by perwulff