libcsp/libcsp #97: CSP_USE_CRC32 macro not working properly

cevans01, 2017-06-19:
I am unable to compile libcsp either with crc32 or without crc32. First, compiling with CRC32:

`$ sudo ./waf configure -vvv --with-os=posix --prefix=/usr/local --enable-bindings --enable-if-can --install-csp --enable-can-socketcan --enable-crc32 build`
`$ nm -D build/libcsp.so | grep crc32`
`$    U csp_crc32_memory `

I get the same result when compiling without --enable-crc32. This will obviously cause any program trying to link libcsp to fail because there is an undefined symbol.

Even stranger, when applying the following patch, which puts a #error as follows:
`#ifdef CSP_USE_CRC32`
`#else`
`#error "err"`
`#endif`

The code fails to compile only when --enable-crc32 is supplied, which is opposite of what I believe it should do, correct?
[error.txt](https://github.com/libcsp/libcsp/files/1085460/error.txt)

perwulff, 2017-06-20:
I think the problem is --enable-can-socketcan expects an argument and assumes --enable-crc32 is its argument.
Try adding =true, --enable-can-socketcan=true

cevans01, 2017-06-20:
Yes, thank you that did fix the issue. I briefly looked into ways to change the wscript to detect if no key-value was passed to --enable-can-socketcan but didn't come up with anything. Thanks again.

closed 2017-06-20 by cevans01