libcsp/libcsp #102: Possible bug in rdp ack retransmissions

Dolmio, 2017-08-16:
If delayed acks is turned off and an ack-packet is lost in transmission csp won't send the ack packet again and that will cause the connection to fail, because the other end will continue transmitting the last received packet over and over again. I think the reason is https://github.com/libcsp/libcsp/blob/17bbef37142d67ebb8bf87044296bf6a79d3a799/src/transport/csp_rdp.c#L406-L412 The ```rdp.rcv_lsa``` and ```rdp.rcv_cur``` will be equal after the first time the ack is sent. Do I understand the code correctly?

johandc, 2017-08-16:
Yes, you are correct. I believe the correct thing would be to always send an ack.

```
if (!conn->rdp.delayed_acks) {
    return true;
}
```

Could i ask you to rebuild and test with this modification and return here with the result. We will target the fix for the upcoming v1.5 release

Dolmio, 2017-08-17:
The that line was changed to the current one in cc1c944b36de1a18602c923833999fcce731e215.
I speculate that the purpose of that was to avoid sending ack every time ```csp_rdp_check_timeouts```is called which is also incorrect behaviour.

johandc, 2017-08-18:
Please try out this ifx.
(note it is based off the v1.5 release branch and not v1.4, if you want to test on a clean v1.4 you can cherry pick the commit onto master)

Dolmio, 2017-08-22:
I get to chance to try it later this week. Sorry for delay.

Dolmio, 2017-08-22:
@johandc: I think your fix would change the ack behaviour so that even if delayed acks are disabled acks would be sent when the ack timeout expires and not only when new packet or duplicate packet is received. I think https://github.com/Dolmio/libcsp/commit/e7ea18124cc9cfc74ca7e0a40fb37f3581b96d46 would keep the old behaviour where acks are not sent if delayed acks is disabled and timeouts expire. I will test it tomorrow.

Dolmio, 2017-08-28:
Any comment on this? In my testing the https://github.com/Dolmio/libcsp/commit/e7ea18124cc9cfc74ca7e0a40fb37f3581b96d46 has worked fine.

johandc, 2017-09-23:
Sorry for the holdup. I will merge this into our release candidate.

PR #114 (2018-01-24): Fix issue where ack is not sent when duplicate packet
arrives and delayed acks is disabled.

Fixes #102
closed 2018-01-24 by johandc