libcsp/libcsp #118: Change release 1.5 to 1.6

perwulff, 2018-08-27:
Because commit #98 breaks a lot of APIs on the branch release-1.5, I would like to propose that we tag commit ca5cc314af26b51db40495c271df273ee9fc53e6 with version 1.5.0 and "rename" the current  release-1.5 branch release-1.6 and move on from there.

The reason is, that GOMspace already uses release-1.5 in some products and we do not want to update these to the latest breaking 1.5 version. BUT we need to do bug-fixes on 1.5. So we would prefer to tag a commit 1.5.0 and make head of release 1.5 -> 1.6.

We hope this is oK with everybody?

johandc, 2018-08-31:
Yes. That is fine with me.

perwulff, 2019-06-20:
Head of release-1.5 pushed to release-1.6. Release-1.6 is moving towards master.

closed 2019-06-20 by perwulff