libcsp/libcsp #120: Interface with PIC 24FJ256GA110 with no operating system

jeremyjallam, 2018-09-11:
I am using a PIC 24FJ256GA110 without an operating system. Is there any reference code I can use as a starting point to write the driver for the GOMSpace transceiver?

Also if someone has advice on how to do this it would be greatly appreciated as I am fairly new to the CSP.

perwulff, 2018-09-19:
hi,
GomSpace has no experiences with running CSP without an OS. We mainly run on 32 bit MCUs with FreeRTOS.
But I think it should be doable by mainly implementing src/arch/* APIs on your platform. There will of course be room for improvements in form of optimisation.

carlgonz, 2018-09-20:
Hi @jeremyjallam. Not sure if you can use LibCSP without an OS. However, for the SUCHAI I nanosatellite we used the PIC24FJ256GA110 with FreeRTOS and LibCSP. Please check the repo and feel free to use the code if it is useful for you: https://github.com/spel-uchile/SUCHAI

johandc, 2018-09-28:
CSP is no longer dependent on tasks since the router is now callable from an event loop.

Semaphores and mutexes are not needed when you run OS less.

The only requirement would be queues, which can very easily be implemented using a ring buffer.

So with a little work, CSP will run fine without an os.

closed 2019-01-25 by johandc