libcsp/libcsp #134: How to build libcsp for STM32F429 MCU on keil MDK-ARM V5.22?

HammiSaif, 2019-12-24:
Hello Everyone
I am trying to build libcsp source code for STM32F429 Cortex-M4 MCU. I am using Keil MDK-ARM V5.22 IDE for my software development. Please tell me how to configure/change wscript (waf script file) to build libcsp for  STM32F429? When I run python waf configure --toolchain=MDK-ARM command I get this message:
"Program ['MDK-ARMgcc'] is not executable"

Why is there gcc at the end of MDK-ARM toolchian?

johandc, 2019-12-24:
WAF is setup to use a GCC style cross compiler. I don't know how to use Keil, but it should be simple to replace the C compiler. Generally I can refer you the WAF documentation, and try a Google with WAF and Keil together.. 

There may be a few issues around some preprocessor defines, and perhaps the use of the GCC __attribute__ tag. I don't know. But if you find any, please report them.

There is also another build system (meson) available unofficially on the space inventor fork. This will be included in next release 1.6, so it will be officially supported too.

HammiSaif, 2019-12-27:
Keil uses ARMCC compiler instead of gcc compiler. ARMCC compiler can not compile and build libcsp. I need to use GNU compiler for ARM to build libcsp. Does it mean I will need to use GNU compiler for ARM for further software development (freeRTOS based) on STM32F429ZTI6 MCU instead of using ARMCC compiler of keil MDK-ARM?

closed 2020-07-05 by johandc