libcsp/libcsp #154: ABI recently broken

anthonyclark-as, 2020-04-21:
I've been following the latest changed in `develop` for a few months and have been testing against many devices that come bundled with some version of `libcsp` - mostly GOMSpace devices.

It appears that the latest large changes (maybe in `csp_conn.c`?) have broken the ABI of `libcsp`. Specifically, a satellite-critical vendor-specific parameter system (actual name omitted but can be discussed more via email) has many features malfunction.

I'm working on diagnosing this issue further and will attach logs here. Is it possible that now is a good time to start implementing common ABI-keeping semantics?

anthonyclark-as, 2020-04-21:
It also seems that `libcsp` v1.4 and older expected `csp_transaction` , `csp_sfp_send`, and `csp_sfp_recv` to be actual symbols rather than `static inline`functions. There might be a way around that with a dummy translation unit, but currently that's not supported. Lots of vendor SDKs that use `libcsp` calls provide `libcsp` sources, but no way of fixing linking errors like these.

johandc, 2020-04-21:
I see your point. We have not been focused on binary compatability between versions. Could you tell us why keeping compatible with the binary interface of v1.4 is important? We are not keeping the API compatible between 1.4, and 1.6. So the ABI will definitely be broken too. (note v1.5 was an internal gomspace release). 

In which build setup (SDK) are you using libcsp? 

In our systems we always include libcsp as a git submodule and build it together with the rest of the firmware image.

johandc, 2020-04-21:
I just realised that the justification here is not CSP being backwards binary compatible in itself. It is because you have some older libraries, from a third party, that you cannot get rebuilt against the latest CSP version.

I don't know which subset of the library your third party binary uses, but if it's just the matter of removing a few inlines I think we might make the performance sacrifice to be backwards compatible.

Otherwise, I really think that in order to progress with CSP, we need to reserve the right to make up for past mistakes and refactor interfaces occasionally. Like once a year with each release.

v1.6 also have a new initialization system, a new interface API and much more. But of course csp_transaction, send, recv and so on we could take better care of not to be breaking.

anthonyclark-as, 2020-04-22:
@johandc Thanks for the reply! I was a bit unsure if binary compatibility between major versions was expected. I see some other "issues" where binary compatibility is mentioned and attempted to be kept. I also really like a majority of the changes in v1.6 - including the much better initialization/interface code. And yes, I am dealing with some pre-compiled libraries that use libcsp v1.4. It is much better to simply submodule libcsp v1.4 to deal with it. I appreciate the reply!

closed 2020-04-22 by anthonyclark-as