libcsp/libcsp #156: Addresses 0 and 31

9a3eedi, 2020-05-05:
Hi,

Are addresses 0 and 31 reserved? I had an e-mail discussion with a CSP node provider, and they casually mentioned that these addresses are reserved. However, the documentation does not talk about that.

johandc, 2020-05-05:
No they are not reserved.

There have been some provisions for broadcast CSP traffic using address 31. But 31 is available if you do not have broadcast enabled. Which I believe nobody uses at the moment.

Address number zero also works fine, we use it as the default for all systems that have not yet received configuration parameters. So in that respect, we choose not to use it actively to avoid conflict when inserting new systems on the bus. 

So you can use both addresses as you please.

9a3eedi, 2020-05-12:
Make sense. It sounds like it's best if I avoid these addresses anyway, just to avoid any conflicts in case we decide to change our design.

closed 2020-05-12 by 9a3eedi