libcsp/libcsp #160: How to import libcsp to existing project

gmardiansyah, 2020-06-28:
Hey, I have project using A3200 board, there is no have libcsp configured on project. Now, I want to add libcsp into project. When I add libcsp into project and try to compile the project there is any error. Can I get a step to import libcsp into my existing project ? 

Thankyou very much!

carlgonz, 2020-06-28:
Hi! Which SDK are you using?
What is the error you are getting?

Kind regards!

*Carlos González Cortés.*
+56975535716

gmardiansyah, 2020-06-29:
> Hi! Which SDK are you using? What is the error you are getting? Kind regards! *Carlos González Cortés.* +56975535716 El dom., 28 de jun. de 2020 a la(s) 09:01, gmardiansyah ( notifications@github.com) escribió:
> […](#)

> Hi! Which SDK are you using? What is the error you are getting? Kind regards! *Carlos González Cortés.* +56975535716 El dom., 28 de jun. de 2020 a la(s) 09:01, gmardiansyah ( notifications@github.com) escribió:
> […](#)

Hi,

Thanks for your reply.

1. I am using A3200 STD 1.0 SDK. 
2. I try to build library file with .a extension (libcsp.a) with command like this : 

waf configure --toolchain=avr32- --with-os=freertos --prefix=install --includes=/home/gmardian/Documents/workspace/lib/libasf /thirdparty/freertos/freertos-8.0.1/Source/include/,/home/gmardian/Documents/workspace/lib/libasf/thirdparty/freertos/freertos-8.0.1/module_config/,/home/gmardian/Documents/workspace/lib/libasf/thirdparty/freertos/freertos-8.0.1/Source/portable/GCC/ARM_CM0/,/home/gmardian/Documents/workspace/lib/libasf/thirdparty/freertos/freertos-8.0.1/Source/portable/GCC/ARM_CM4F/

3. Configure command is success. But when I run command to build like this : 

waf install build

4. I got error like this : 

In file included from ../src/csp_conn.h:26,
                 from ../src/csp_conn.c:21:
../include/csp/arch/csp_semaphore.h:106: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'csp_bin_sem_handle_t'
../include/csp/arch/csp_semaphore.h:107: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'csp_mutex_t'
../include/csp/arch/csp_semaphore.h:131: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:139: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:148: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:156: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:168: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:176: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:185: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:193: error: expected ')' before '*' token
../include/csp/arch/csp_semaphore.h:202: error: expected ')' before '*' token
In file included from ../src/csp_conn.c:21:

Any other way to import library  into my project ? 

Thankyou very much!
GM

carlgonz, 2020-06-29:
It seems to be a build error related with the specific GomSpace A3200 SDK not with libcsp itself.
What I do is to compile my code using the GomsSpace SDK build system. In other platforms I build the library and integrate it in my build system.

You should try to write GomSpace for support. I have some experience programming the Nanomind A3200, so you can send me an e-mail to carlgonz@uchile.cl to see if I can help you.

Kind regards!

gmardiansyah, 2020-06-30:
Hi, 
I'm not sure if problem is on A3200 SDK, because when I try to compile SDK without libcsp, its successfully. Thanknfor your email information.

Thanks

gmardiansyah, 2020-07-02:
> It seems to be a build error related with the specific GomSpace A3200 SDK not with libcsp itself.
> What I do is to compile my code using the GomSpace SDK build system. In other platforms I build the library and integrate it in my build system.
> 
> You should try to write GomSpace for support. I have some experience programming the Nanomind A3200, so you can send me an e-mail to carlgonz@uchile.cl to see if I can help you.
> 
> Kind regards!

Hi, I am not sure the problem on SDK, because if I compile SDK standalone without libcsp, build process is success. 

Thanks.

gmardiansyah, 2020-07-03:
This issue is closed.

closed 2020-07-03 by gmardiansyah