libcsp/libcsp #162: XTEA encrypt packet nonce too predictable

diamondo25, 2020-06-29:
The csp_xtea_encrypt_packet function uses a 2-byte signed value (`rand()`) as part of the 8 byte nonce, which results into the following key (in Little Endian):

`XX XX 00 00 01 00 00 00`
where XX XX is a value between 0 and `RAND_MAX` so 32767. Additionally, `srand()` is only called by `csp_conn_init`, so I cannot guarantee its actually pseudorandom. The second part is the GCM counter part, so that's fine, I suppose. I would still recommend seeding both the first and second two bytes with rand(), although that might not solve the issue completely. Maybe a save interface for rand() that supports hardware RNG would help? Eventually you would want to try to reduce the amount of plaintexts encrypted with the same key.

https://github.com/libcsp/libcsp/blob/87006959696c78f70535ab382b0bcd4cb5a6558d/src/crypto/csp_xtea.c#L128

johandc, 2020-07-05:
Okay, thanks for pointing this out.

The built-in encryption has not really gotten a lot af attention since it's implementation a Lot of years ago. It is my impression that it is not used at all.

We are working on CSP 2.0 right now. So I am thinking about how we can do this better.

Typically satellites does not want to encryption specific connections. (Right now this is a socket option). So I think it would be more useful if it was a lower layer option, that could be enabled on the radio. So all the traffic over an interface could be encrypted. 

Do you use XTEA actively, and do you have any comments or ideas for how to do this better in 2.0

johandc, 2021-06-02:
I believe hmac and xtea will be deprecated from  CSP 2.0 and forwards. People should move to better encryption standards such as AES or libNaCl.

closed 2021-06-02 by johandc