libcsp/libcsp #163: Interrupted system call in the csp_if_zmqhub interface

hughbagan, 2020-06-30:
Hello. I've got a runtime error in our libcsp project that uses the ZMQ interface.

While running `libcsp/build/zmqproxy`, it will randomly start barfing out this OS system call error every iteration:

```
1593475542.247350 Starting ZMQproxy
Running at 8
Timer Resolution for Run TimeStats is 100 ticks per second.

...

1593475474.424639 RX ZMQHUB: Interrupted system call
1593475474.425606 RX ZMQHUB: Interrupted system call
1593475474.426603 RX ZMQHUB: Interrupted system call
... (etc)
```

Sometimes it will do this immediately upon starting the program, other times it will start after a minute. 

The error message is, of course, originating here: https://github.com/libcsp/libcsp/blob/master/src/interfaces/csp_if_zmqhub.c#L88 

We think that the cause might be because the ZMQ interface code is not thread-safe. What do you think? Could this actually a problem with ZMQ and not the interface? Bearing in mind that we have libcsp built for POSIX.

Any insight you could provide would be valuable.  
Thank you!

---

CSP version: latest master at the time of writing (https://github.com/libcsp/libcsp/commit/87006959696c78f70535ab382b0bcd4cb5a6558d)  
Our project:  
- https://github.com/AlbertaSat/ex2_services (see the Dockerfile for the zmqproxy command)
- https://github.com/AlbertaSat/SatelliteSim

hughbagan, 2020-07-08:
I think I've narrowed down the problem such that it's probably an issue with our code and not CSP. Still don't know the cause, though.

johandc, 2020-07-08:
Okay. Please report if/when we can close the issue.

arrooney, 2020-07-16:
Update: this issue is not our code, the issue can be replicated with the example client/server code, and is due to some interrupt occuring during a system call made by ZMQ. It's likely that something in this code isn't quite thread-safe. If we could have a hand looking into this, it would be greatly appreciated!
Otherwise, if anyone on your team can make the code work without seeing this problem, i'd be curious what system they're running.

johandc, 2020-07-17:
What platform are you running it on?

arrooney, 2020-07-20:
We're using CSP compiled for unix  - we've seen in on Arch and Ubuntu. We're making calls to CSP from a FreeRTOS simulated environment, which essentially wraps FreeRTOS to unix calls.

johandc, 2020-07-23:
It sounds like the issue may be in the FreeRTOS wrapper layer then. Because running CSP native compiled for POSIX, does not produce these debug messages

johandc, 2021-06-02:
Did you solve this?

We use zmqhub extensively on our Linux devices. And have kept an eye on this and could not reproduce.

Closing by timeout. Please reopen if still an issue.

closed 2021-06-02 by johandc

arrooney, 2021-06-02:
> Did you solve this?
> 
> We use zmqhub extensively on our Linux devices. And have kept an eye on this and could not reproduce.
> 
> Closing by timeout. Please reopen if still an issue.

No, we never found the root cause of this - it must have been somewhere else in our overly-complex system (probably the FreeRTOS simulator). We ended up using the FIFO interface from the old version of CSP and gave up on this. Issue is okay to close. Thanks!

johandc, 2021-06-02:
You are welcome. Keep an eye out for awesome CSP 2.0 coming soon.