libcsp/libcsp #164: Static (stack-only) memory management

pwnorbitals, 2020-06-30:
As of now and if I understand well, libCSP cannot be used on platforms where heap-based `malloc`-type allocation is impossible. Seeing how small the memory footprint of the library is, I'm confident that removing this limitation is very possible. 

The documentation in `doc/memory.rst` explains that `Future versions of libcsp may provide a pure static memory layout`. Has there been any progress with it lately ?

johandc, 2020-07-05:
Yes. We are currently working on CSP 2.0, which may very well come with support for fully static allocation.

Do you have any time to work on it yourself, you are welcome to come up with a suggestion.

pwnorbitals, 2020-07-10:
I will do my best. Should I work on the current develop branch ?

johandc, 2020-07-10:
Yes, thank you.

I think it would be nice if the buffer space could be passed to CSP through the config_t on the csp_init call.

johandc, 2021-06-02:
This issue Will be solved before the 2.0 release

yashi, 2021-07-25:
Would you mind sharing your idea of how you resolve? I'd love to use static memory instead of heap memory API.

It'd be nice if we can use predefined data section memory as object store.
e.g) https://docs.zephyrproject.org/latest/reference/kernel/data_passing/message_queues.html

I think this is similar to [`xQueueCreateStatic()`](https://www.freertos.org/xQueueCreateStatic.html) of FreeRTOS.  I couldn't find [`K_MSGQ_DEFINE()`](https://docs.zephyrproject.org/latest/reference/kernel/data_passing/message_queues.html#c.K_MSGQ_DEFINE) like macro in FreeRTOS.

johandc, 2021-07-25:
Basically I would pass in, pointers to memory areas for the packet buffer pool etc. Through additional fields in the CSP init data structure. If they are not set I would keep the dynamic allocation.

johandc, 2021-09-19:
Fixed on latest 2.0 branch of spaceinventor you are welcome to test.
completely removed the csp_malloc system while i was at it. Since it's not needed internally, i don't think it's CSP's responsibility to provide wrapers for malloc any longer.

closed 2021-09-19 by johandc