libcsp/libcsp #166: CRC32 endianness incorrect

binglebear, 2020-08-05:
One issue I've found with the CRC32C calculation is the writing the endian is wrong for the lookup table used (there are forward and reverse tables). CRC32 has a property that if you calculate the crc32(data)=x and then concatenate x to the data, then the resultant crc32(data,crc) is a static value - a value derived from the polynomials used and if it is a forward or backwards implementation.

For example a 100 byte ping 0..99 resulting in crc(data) = 0xc1caebe5 results in {0xc1 0xca 0xeb 0xe5} being added to .data
Changing the crc32 endian from big to little endian (and hence adding {0xe5 0xeb 0xca 0xc1} to .data) and instead calc crc32(data,crc) = 0x48674bc7 (for crc32c)
Modifying the ping data to 1..100, and writing the crc as little endian, the crc32(data+crc) is **also** 0x48674bc7 even though the data is different. As long as the crc32c is written in little endian, the crc of (data,crc) will ALWAYS result in 0x48674bc7, assuming no corruption - this significantly simplifies the checking of data as you don't need to do crc32(data(1..length-4)), swap/copy the crc32 and then compare the values.

Ethernet uses CRC32(a) in this way, the drivers simple check for the static value and deduce 4 off the length and then process the frame. I discovered this long after I had written a FCS engine!!

This property crucially improves latency. The CRC32 calcs can be done at the irq/driver level, kiss uart for instance can calculate the crc32 inside the irq routine (one byte at a time so quick enough), and on receipt of the FEND can simple check the CRC32==0x48674bc7. Instead libcsp needs to exclude the crc bytes so hence take a latency hit while small microcontrollers chug through the whole frame as it doesn't know which byte is the start of the crc32 until it knows the end.

Perhaps this can be corrected in v2.0 as it would break existing systems using the existing endian.

johandc, 2020-08-05:
Yes. This sounds exactly like something we want to look at for 2.0. I haven't heard of it before. So need to do some research. It almost sounds too good. 😃

Can you provide a reference implemention? Would it just be a swap of endian? And if so, will this performance benefit only exist on little endian systems?

binglebear, 2020-08-06:
Yes, it's just changing to little endian of the crc32 value as written. I'll see what I can demo on the weekend - I haven't done push requests so maybe a change to tinker on more advanced git :-)

johandc, 2021-06-02:
I don't think this will make it into CSP 2.0 we will remain on the old checksum. The reason is there's already constellations out there that runs on the development version. So I don't want to break them for a slight speed improvement.

closed 2021-06-02 by johandc