libcsp/libcsp #168: Better way of passing CAN ID and Mask to the driver

9a3eedi, 2020-08-27:
In my project, I had previously written a custom CAN driver for CSP 1.4 for an embedded FreeRTOS system, I had to implement a few functions, one of which was can_init. CSP would call this function to initialize the CAN device, and passes in a ready-made CAN ID and a mask, and I can use that to set up my CAN device. This was straightforward to work with.

I decided to upgrade my project to CSP 1.6. With CSP 1.6, I had to rewrite the driver code to wrok with the new way of doing things. WLooking at the socketcan driver code as an example, it seems that the driver code needs to figure out on its own what the CAN address and mask should be. This is what's found in the socketcan code:

`
		struct can_filter filter = {.can_id = CFP_MAKE_DST(csp_get_address()),
						.can_mask = CFP_MAKE_DST((1 << CFP_HOST_SIZE) - 1)};`

This is a minor thing.. but in my opinion it's not as straightforward for the driver developer as the previous method. it requires the driver to understand too much about the CFP macros. It would be better if there were at least some cleaner way of defining to the driver what the address and mask should be, as it were before. 

Just feedback from a custom driver developer.

johandc, 2020-08-28:
You are right, this information should come from the layer above the can driver. And in fact, in v2.0 where we changed the CFP format, there is now an ugly if-case inside our can_driver_init code, trying to figure out whether to generate a version 1 or 2 CFP mask. I would like to see this change, and perhaps include more drivers into the library as examples. For example include popular embedded frameworks, such as atmel start, or STM embedded libraries. What platform did you write a driver for?

9a3eedi, 2020-09-01:
I was developing a driver for an MCP2515 being driven by SPI by an EFM32GG.

Interesting, I wasn't aware there are now two versions of CFP. Perhaps I should do more reading.

johandc, 2021-06-02:
We need this for broadcast to work too.

johandc, 2021-11-24:
I will upload an example can driver to contrib. Which should be helpfull for driver development.

commit d4326272 (2021-11-24): contrib: can driver: atsamc21

Fixes #168
closed 2021-11-24 by johandc