libcsp/libcsp #169: python binding

schaefer01, 2020-10-01:
HI,
  I built and tried the c programs, which worked but then tried the python version, and got an exception.

$ python3 python_bindings_example_client.py -z ZMQ
  File "python_bindings_example_client.py", line 54, in <module>
    print("CMP ident:", libcsp.cmp_ident(options.server_address))
csp.Error: csp_cmp_ident(), result/error: -3

I looked inside pycsp.c and found a mapping  from csp_cmd_ident to cmp_ident
I then grep'd the source for cmp_ident, and found the function do_cmp_ident, but found no existence for cmp_ident,
so I'm thinking either a copy-and-paste error, or the function name changed but the python binding did not.

johandc, 2020-10-03:
The cmp functions are implemented using macros. So there is no direct mapping to cmp_ident. I believe the macros call the same function, do_cmp_send or something to that extent, with an argument for the IDENT type.

schaefer01, 2020-10-05:
thank you, I figured the problem out, it was operator (me) error, if the client connects before there is a server, then an error is raised.

closed 2020-10-05 by schaefer01