libcsp/libcsp #170: unable to use can interface on STM32f405

AgentK47, 2020-10-04:
Hi,I am implementing CSP on STM32F405 I did the loop back test and the server and client example worked. But I spent a lot of time trying to add CAN interface but no success. 
I was able to send data to the other board but the rx is not working. There's an example on how to use CAN on linux but not for a microcontroller. Please would you be able to provide some examples. Here is my implementation for the rx. 

void CAN1_RX0_IRQHandler(void)
{
	can_context_t * ctx = calloc(1, sizeof(*ctx));
	
	HAL_CAN_IRQHandler(&hcan1);
	static portBASE_TYPE xTaskWoken = pdFALSE;
	HAL_CAN_GetRxMessage(&hcan1, CAN_RX_FIFO0, &pRxHeader, &r);
	printf("%s\n",(char*)r);

	/* Call RX callbacsp_can_rx_frameck */
	csp_can_rx(&ctx->iface, pRxHeader.ExtId, &r, pRxHeader.DLC, &xTaskWoken);

}
Regards, Kevin

johandc, 2020-10-04:
First of all. The can context is supposed to be a globally shared object. So do not overwrite it for each frame.

Secondly, you should never ever call any dynamic allocation from interrupt context.

Third, you must be watchful when using printf in interrupt context. Most implementations of printf is not interrupt safe. And your UART should also be interrupt safe.

Other than that, your implementation looks okay. I cannot see what r is. Usually it is bad practice to use one letter variables for anything else than loop incrementors. But I assume it is a buffer pointer.

johandc, 2020-10-04:
```
#include <stdio.h>
#include <driver_init.h>
#include <hal_can_async.h>

#include <FreeRTOS.h>
#include <semphr.h>

#include <csp/csp.h>
#include <csp/interfaces/csp_if_can.h>

#include <drivers/csp_driver_can.h>
#include <drivers/sn65hvd230.h>

/** Driver configration */
static struct mcan_s {
	can_mode_e mode;
	uint32_t id;
	uint32_t mask;
	csp_can_interface_data_t ifdata;
	SemaphoreHandle_t lock;
	StaticSemaphore_t lock_buf;
	csp_iface_t interface;
} mcan[1] = {
	{
		.mode = 0,
		.lock = NULL,
		.interface = {
			.name = "CAN",
			.interface_data = &mcan[0].ifdata,
			.driver_data = &mcan[0],
		},
	}
};

void CAN_0_rx_callback(struct can_async_descriptor *const descr) {

    int xTaskWoken = pdFALSE;
	struct can_message msg;
	uint8_t data[8];
	msg.data = data;

	while(can_async_read(descr, &msg) == ERR_NONE) {

		//csp_hex_dump("rx", msg.data, msg.len);

		/* Process frame within ISR
		 * (This can also be deferred to a task with: csp_can_process_frame_deferred) */
		csp_can_rx(&mcan[0].interface, msg.id, msg.data, msg.len, &xTaskWoken);

	}

	portYIELD_FROM_ISR(xTaskWoken);

	return;
}

void CAN_0_irq_callback(struct _can_async_device *dev, enum can_async_interrupt_type type) {
	can_async_disable(&CAN_0);
	mcan[0].interface.tx_error++;
	can_async_enable(&CAN_0);
	return;
}

int csp_can_tx_frame(void *driver_data, uint32_t id, const uint8_t * data, uint8_t dlc) {

	struct mcan_s * driver = (struct mcan_s *)driver_data;

	if ((driver == NULL) || (driver->lock == NULL)) {
		return 0;
	}

	struct can_message msg;
	msg.id = id;
	msg.type = CAN_TYPE_DATA;
	msg.data = data;
	msg.len = dlc;
	msg.fmt  = CAN_FMT_EXTID;

	/* Task locking */
	while(xSemaphoreTake(driver->lock, 1) == pdFALSE);

	/**
	 * Blocking IO:
	 *
	 * With CAN blocking IO is needed because we only have room for 32 frames or 256 bytes.
	 * This is barely enough to hold a CSP packet if large MTU is used.
	 * We could go for async transmission with software queues, this will allow for larger amounts
	 * of data to be queued. However in this case, simplicity is chosen over performance.
	 * When the TX FIFO is full, we ask the task to sleep a tick.
	 * Usually data heavy fuctions are running in their separate tasks anyways.
	 * A CAN frame is 96 bits and is transmitted within 96 us.
	 * A tick period can vary from 1 to 10 ms (typically)
	 * 3 retries have been chosen because a futher dealy than this would almost certainly be
	 * due to an error.
	 */
	int attempts = 3;
	while((attempts > 0) && (can_async_write(&CAN_0, &msg) != ERR_NONE)) {
		vTaskDelay(1);
		attempts--;
	}

	/* Unlock */
	xSemaphoreGive(driver->lock);

	if (attempts > 0) {
		return CSP_ERR_NONE;
	} else {
		return CSP_ERR_BUSY;
	}
}

csp_iface_t * csp_driver_can_init(int id, can_mode_e mode, uint32_t bitrate)
{
	/* Create a mutex type semaphore. */
	mcan[id].lock = xSemaphoreCreateMutexStatic(&mcan[id].lock_buf);

	const csp_conf_t *csp_conf = csp_get_conf();

	/* Generate ID */
	uint32_t can_mask = 0;
	if (csp_conf->version == 2) {
		mcan[id].id = csp_get_address() << CFP2_DST_OFFSET;
		can_mask = CFP2_DST_MASK << CFP2_DST_OFFSET;
	} else {
		mcan[id].id = CFP_MAKE_DST(csp_get_address());
		can_mask = CFP_MAKE_DST((1 << CFP_HOST_SIZE) - 1);
	}

	if (mode == CSP_CAN_MASKED) {
		mcan[id].mask = can_mask;
	} else if (mode == CSP_CAN_PROMISC) {
		mcan[id].mask = 0;
	} else {
		csp_log_error("Unknown CAN mode");
		return NULL;
	}

	/* The MTU is configured run-time, since the buffer size can be configured externally
	 * however, it must not exceed 2042 due to the CFP_REMAIN field limitation
	 * CFP_REMAIN gives possibility of 255 * 8 bytes = 2040
	 * CSP_BEGIN frame, has two additional bytes, in total 2042 */
	mcan[id].interface.mtu = csp_conf->buffer_data_size;
	if (mcan[id].interface.mtu > 2042) {
		mcan[id].interface.mtu = 2042;
	}

	mcan[id].ifdata.tx_func = csp_can_tx_frame;
	mcan[id].interface.interface_data = &mcan[id].ifdata;

	/* Regsiter interface */
	csp_can_add_interface(&mcan[id].interface);
	//csp_iflist_add(&mcan[id].interface);

	can_async_register_callback(&CAN_0, CAN_ASYNC_RX_CB, (FUNC_PTR)CAN_0_rx_callback);
	can_async_register_callback(&CAN_0, CAN_ASYNC_IRQ_CB, (FUNC_PTR)CAN_0_irq_callback);
	can_async_enable(&CAN_0);

	struct can_filter filter;
	filter.id   = mcan[id].id;
	filter.mask = mcan[id].mask;
	can_async_set_filter(&CAN_0, 0, CAN_FMT_EXTID, &filter);

	return &mcan[id].interface;

}
```

closed 2020-10-04 by johandc

AgentK47, 2020-10-07:
Thanks for replying as you suggested I declared can context globally. And corrected some few remarks, while debugging I can see the information in the buffer transmitted to csp_can_rx.

AgentK47, 2020-10-07:
struct can_driver{
	char name[CSP_IFLIST_NAME_MAX + 1];
	csp_iface_t iface;
	csp_can_interface_data_t ifdata;

};

struct can_driver ctx={

		.iface= {
				.name = "CAN",
				.interface_data = &ctx.ifdata,
				.driver_data = &ctx,
			},

};

/**
  * @brief This function handles CAN1 RX0 interrupts.
  */
void CAN1_RX0_IRQHandler(void)
{

	HAL_CAN_IRQHandler(&hcan1);
	static portBASE_TYPE xTaskWoken = pdFALSE;

	HAL_CAN_GetRxMessage(&hcan1, CAN_RX_FIFO0, &pRxHeader, &rx);

	/* Call RX callbacsp_can_rx_frameck */
	csp_can_rx(&ctx.iface, pRxHeader.ExtId, rx, pRxHeader.DLC, &xTaskWoken);

}

static int csp_can_tx_frame(void * driver_data, uint32_t id, const uint8_t * data, uint8_t dlc)
{
	if (dlc > 8) {
		return CSP_ERR_INVAL;
	}
		pHeader.DLC=dlc;
		pHeader.IDE=CAN_ID_STD;
		pHeader.RTR=CAN_RTR_DATA;
		pHeader.StdId=0;
		pHeader.ExtId=id;

		for (uint8_t i = 0; i < dlc; i++)
			{
				tx[i] = data[i];
			}
		HAL_StatusTypeDef rep;
			do
			{
				rep=HAL_CAN_AddTxMessage(&hcan1, &pHeader, tx, &pTxMailbox);
//				HAL_Delay(10);
			}
			while((rep=HAL_OK));

	return CSP_ERR_NONE;
}

int csp_can_stm32_init(const char * device, const char * ifname,  csp_iface_t ** return_iface)
{
	if (ifname == NULL) {
		ifname = CSP_IF_CAN_DEFAULT_NAME;
	}

	csp_log_info("INIT %s: device: [%s], bitrate: %d, promisc: %d",
	ifname, device, bitrate, promisc);
	sFilterConfig.FilterFIFOAssignment=CAN_FILTER_FIFO0;
	sFilterConfig.FilterIdHigh=0;
	sFilterConfig.FilterIdLow=0;
	sFilterConfig.FilterMaskIdHigh=0;
	sFilterConfig.FilterMaskIdLow=0;
	sFilterConfig.FilterScale=CAN_FILTERSCALE_32BIT;
	sFilterConfig.FilterActivation=ENABLE;
	HAL_CAN_ConfigFilter(&hcan1, &sFilterConfig);

	HAL_CAN_Start(&hcan1);
	HAL_CAN_ActivateNotification(&hcan1, CAN_IT_RX_FIFO0_MSG_PENDING);

	strncpy(ctx.name, ifname, sizeof(ctx.name) - 1);
	ctx.iface.name = ctx.name;
	ctx.iface.interface_data = &ctx.ifdata;
	ctx.iface.driver_data = &ctx;
	ctx.ifdata.tx_func = csp_can_tx_frame;

	/* Add interface to CSP */
        int res = csp_can_add_interface(&ctx.iface);
	if (res != CSP_ERR_NONE) {
		csp_log_error("%s[%s]: csp_can_add_interface() failed, error: %d", __FUNCTION__, ctx.name, res);

		return res;
	}

	if (return_iface) {
		*return_iface = &ctx.iface;
	}

	return CSP_ERR_NONE;
}

here is my code maybe I'm missing something. I appreciate your help :)

johandc, 2020-10-07:
So you didn't say what your issue is?

AgentK47, 2020-10-07:
Sorry the packet is not received by the server. csp_can_rx is called with all the required variables.

johandc, 2020-10-08:
Try to enable debugging CSP_PACKET and see if it is delivered to the router task. If not then start debugging on the defamer by inserting printf till you figure out what is wrong with the data

AgentK47, 2020-10-09:
I think I found the issue but I don't know how to resolve it.

In csp_if_can.c  this line is never called "	csp_qfifo_write(buf->packet, iface, task_woken);". The reason is because this condition is always true :		/* Check if more data is expected */ if (buf->rx_count != buf->packet->length) break;
This "CFP_TYPE(id)" always return 0.

Thank you.

johandc, 2020-10-11:
It sounds like your CAN id may be misplaced in the 32 bit field. Try printing out the IDs and check for the begin and end flags. Then check with the appropriate bit mask of CFP TYPE. Some hardware platforms represent the ID fields in different ways. The way CFP expects it is right shifted into a 32 bit value.

AgentK47, 2020-10-12:
It's working. Thanks

AgentK47, 2020-10-23:
Interfaces
LOOP       tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (B) rxb: 0 (B) MTU: 0

I2C        tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (B) rxb: 0 (B) MTU: 256

CAN        tx: 00000 rx: 00000 txe: 00000 rxe: 00000
           drop: 00000 autherr: 00000 frame: 00000
           txb: 0 (B) rxb: 0 (B) MTU: 2042

Route table
1/5 LOOP
0/0 I2C

I don't why I can't use CAN and I2C at the same time. Rooting Table is only showing I2C. 
I'm rooting 1 @ to i2C and 2 to CAN. But they are all using I2C for some reason.

Thanks