libcsp/libcsp #171: RDP Header size isn't accessible from public headers

9a3eedi, 2020-10-07:
Say I made a CSP network wih an MTU of 256. When defining my structures, I need to take into account the additional headers that get appended to the data section of a CSP packet, such as CRC, HMAC, etc. So if I enable CRC and HMAC, the resulting MTU is (256-4-4). For CRC and HMAC, it's well understood how big those headers are, but for RDP, there's also a header that is appended to every RDP packet, but I can only find out the size of this header by inspecting the code in csp_rdp.c, which isn't very nice from a design point of view. 

Generally speaking, there should be a way of knowing programmatically how much additional data will be added to packets of a particular CSP connection, so that the sizes of data inside a packet can be known better... but there is no way of finding out in the current design, unless I missed something.

Or am I thinking about this the wrong way? Should the MTU for a network be defined independent of the additional data that gets added to every packet?

johandc, 2020-10-07:
To add further to this issue. The radio link typically has very different MTU based on the amount of FEC that is added to a packet. But generally it becomes a mess to guesstimate how much higher layers can add to the packet payload before reaching the interface level. It has also been further complicated by the addition of 2 bytes in the CSP 2 header.

I don't have a real solution to this. And in fact most IP network systems also suffers from this problem. In this case the end to end MTU is found by setting the don't fragment flag and sending progressively larger IP packets untill it gets dropped.

But in general I think the MTU of the interface should be set to the max size of the packet length field. So this excludes the length of the CSP header which can now also vary between CSP 1 and 2 header. But since the CRC32, and the RDP, XTEA or HMAC portion are added at the end of the data field these bytes should be included in the MTU.

Meaning that if you use any of those features you as a user should manually adjust the max size you are sending out accordingly.

9a3eedi, 2020-10-09:
I agree that figuring all these numbers out is rather tricky. I just feel that this information could be better documented, either in documentation, or in code. Another idea could be to add an API function where you pass in an csp_conn_t and based on the connection properties, it will return the number of bytes of padding that will be added to the packet.

johandc, 2020-10-09:
Well, it also comes down to which other systems your CSP packets has to travel through on its way to the destination. The maximum transfer unit It cannot purely be calculated locally, it must be tested through the network to be sure. The interface MTU is more of an absolute limit for that particular interface, it does no tell what the other interface MTU's are.

johandc, 2021-06-02:
We can add the rdp-header size to the header-file.

johandc, 2021-10-31:
Added 
```
#define CSP_RDP_HEADER_SIZE 5
```
to csp_types.h

closed 2021-10-31 by johandc