libcsp/libcsp #173: Different CSP Header using ZMq (python) and CAN (C)

IlanBehar, 2020-10-27:
Hello,

I'm working with the libcsp and until recently I was using a CAN bus and coded my communication in C.
Everything works perfectly and I can easily dump the trafic and by doing so I see that before my payload there is a 6 bytes : the 4 bytes of the CSP header and 2 for the length of the payload. 

Then I recently wanted to update one of my nodes so it could use ZMQ through the python bindings. 
Doing so I end up with a 5 bytes header that doesn't correspond at all to the header I had when using CSP over CAN. 

Is this a bug or is it an expected behaviour ? And if it is an expected behaviour is there a documentation that explains the header when using ZMQ ?

johandc, 2020-10-29:
The representation of a CSP packet of course will vary between different transfer mediums.

We don't have much documentation, but try to keep the code simple. 

You can fairly easily see in the implementation of the ZMQ interface that it consists of a filter byte and the CSP header.

9a3eedi, 2021-05-30:
While working on ZMQ, we've noticed the same thing. The first byte corresponds to the via address. This is fine. The 4 bytes after it is supposed to be the 4-byte CSP ID. We noticed that the ID is transmitted in little endian, which didn't seem right since on every other interface (including CAN) it was transmitted in big endian. Upon further inspection, we saw that the ID was transmitted in host byte order

From csp_zmqhub_tx() in csp_if_zmqhub.c:
```
	uint16_t length = packet->length;
	uint8_t * destptr = ((uint8_t *) &packet->id) - sizeof(dest);
	memcpy(destptr, &dest, sizeof(dest));
	csp_bin_sem_wait(&drv->tx_wait, 1000); /* Using ZMQ in thread safe manner*/
	int result = zmq_send(drv->publisher, destptr, length + sizeof(packet->id) + sizeof(dest), 0);
```

And it was received in host byte order. From csp_zmqhub_task:

```
	// Copy the data from zmq to csp
	const uint8_t * rx_data = zmq_msg_data(&msg);

	// First byte is the "via" address
	++rx_data;
	--datalen;

	// Remaining is CSP header and payload
	memcpy(&packet->id, rx_data, datalen);
	packet->length = (datalen - sizeof(packet->id));

	// Route packet
	csp_qfifo_write(packet, &drv->iface, NULL);
```

In my case, both the sending and the receiving machine were little-endian, so in the end it all worked out. But this will fail if one of the machines was big endian. It also threw us off when reading the byte data.

I think the correct behaviour is to covert the ID to network byte order before sending. This is consistent with the behaviour found in other interfaces, e.g.:

CAN (RX):
```
		/* Copy CSP identifier (header) */
		memcpy(&(buf->packet->id), data, sizeof(buf->packet->id));
		buf->packet->id.ext = csp_ntoh32(buf->packet->id.ext);
```
CAN (TX):
```
        /* Copy CSP headers and data */
	const uint32_t csp_id_be = csp_hton32(packet->id.ext);
	const uint16_t csp_length_be = csp_hton16(packet->length);

	uint8_t frame_buf[MAX_BYTES_IN_CAN_FRAME];
	memcpy(frame_buf, &csp_id_be, sizeof(csp_id_be));
	memcpy(frame_buf + sizeof(csp_id_be), &csp_length_be, sizeof(csp_length_be));
	memcpy(frame_buf + CFP_OVERHEAD, packet->data, bytes);
```

KISS (TX):
```
	/* Save the outgoing id in the buffer */
	packet->id.ext = csp_hton32(packet->id.ext);
	packet->length += sizeof(packet->id.ext);
```

KISS (RX):
```
	/* Convert the packet from network to host order */
	ifdata->rx_packet->id.ext = csp_ntoh32(ifdata->rx_packet->id.ext);
```

What also threw me off is the way the IDs are represented in csp_types.h
```
typedef union {
    /** Entire identifier. */
    uint32_t ext;
    /** Individual fields. */
    struct __attribute__((__packed__)) {
#if (CSP_BIG_ENDIAN || __DOXYGEN__)
        unsigned int pri   : CSP_ID_PRIO_SIZE;  //!< Priority
        unsigned int src   : CSP_ID_HOST_SIZE;  //!< Source address
        unsigned int dst   : CSP_ID_HOST_SIZE;  //!< Destination address
        unsigned int dport : CSP_ID_PORT_SIZE;  //!< Destination port
        unsigned int sport : CSP_ID_PORT_SIZE;  //!< Source port
        unsigned int flags : CSP_ID_FLAGS_SIZE; //!< Flags, see @ref CSP_HEADER_FLAGS
#elif (CSP_LITTLE_ENDIAN)
        unsigned int flags : CSP_ID_FLAGS_SIZE;
        unsigned int sport : CSP_ID_PORT_SIZE;
        unsigned int dport : CSP_ID_PORT_SIZE;
        unsigned int dst   : CSP_ID_HOST_SIZE;
        unsigned int src   : CSP_ID_HOST_SIZE;
        unsigned int pri   : CSP_ID_PRIO_SIZE;
#endif
    };
} csp_id_t;
```

My initial understanding is that the defines are done to ensure that the ID is always represented in big endian in memory, as it is presumed that the compiler will reverse the bit order when compiling on a little endian machine. This means that conversions will not be necessary when copying the ID's as bytes from memory. But this is a dangerous assumption because the way that the compiler decides to put the data in memory is implementation specific.

I've written a test program to test this assumption [here](https://gist.github.com/9a3eedi/4a1b2538d1fc0d7fba2a45e6170d40a4), and I've attached the C code here. I've defined the ID union similar to how it would be defined in a little-endian system. I expect that reading the value as an int should be identical to reading it as a byte array, since in the end it should be in big-endian. I am using gcc 7.5.0 on Ubuntu 18.04 64-bit. 

The following is the output.

```
ext = 0xA011F208
bytes = 8 F2 11 A0
```

Perhaps this should be a separate issue, but this might be why you're seeing it completely different than what you expect. However, I might be wrong here, maybe I'm missing something. Please let me know if there's any mistake.

johandc, 2021-06-02:
Hello.

Thanks for the very extensive analysis of this. Indeed the zmqhub interface is host order, which is a bug. Which is not possible to solve in 1.6 because it would break existing working installations.

I am tagging this for the 2.0 release.

And btw. your second remark about the packed bit fields have been solved already on the development branch.

https://github.com/libcsp/libcsp/blob/develop/src/csp_id.c

johandc, 2021-10-31:
This is solved in CSP 2.0

closed 2021-10-31 by johandc