libcsp/libcsp #174: examples, local variable doesn't get to global space where it should be used

schaefer01, 2020-10-28:
HI,
  I am starting with your client/server c example, and porting to a kiss uart.

 your example code calls csp_usart_open_and_add_kiss_interface() to add the interface but does not pass the configuration variable "conf" to the global that will  use it, "csp_conf" so its not really added.

How I figured this out was, I ran the code and the loopback driver was called, not the kiss driver, I then traced the code
  to the bait, i.e. "conf" called after csp_init() that remains local while there is a switch as csp_init()'s parameters copied to  "csp_conf" will still be used. Very clever, very subtle.

Am I wrong? And if not wrong, then any help?

schaefer01, 2020-11-02:
not quite true after I added diagnostics, it appears the drivers are added to a list, the first driver is loopback.
How do I pick the driver out I want (not the loopback driver) before it is used?

johandc, 2020-11-03:
You use the routing table to specify which interface should be used to send to which destinations.

schaefer01, 2020-11-03:
Hi Johan,

The default address added to the routing table is loopback set via csp_init(), and selecting the uart via the command line also
seems to add the same address to the routing table. So, the example adds two addresses to the routing table, both routes the same.

in csp_init():

csp_route_set(csp_conf.address, &csp_if_lo, CSP_NO_VIA_ADDRESS);

/* Also register loopback as default, until user redefines default route */

csp_route_set(CSP_DEFAULT_ROUTE, &csp_if_lo, CSP_NO_VIA_ADDRESS);

in csp_server_client.c

else if (default_iface) {
        csp_rtable_set(CSP_DEFAULT_ROUTE, 0, default_iface, CSP_NO_VIA_ADDRESS);

There appears no key-collision check to prevent adding two identical addresses with different intent.

The address selected for ping (also in the example) finds the first entry in the routing table, which is loopback.

Are there any non-trivial examples for using the routing table for setting destinations?

bob s.

schaefer01, 2020-11-03:
of course I may be wrong, running the code again, I see two different addresses, but I’m not sure how the loopback
  address gets into the ping call.

Route table
1/5 LOOP
0/0 KISS

On Nov 3, 2020, at 11:03 AM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

Hi Johan,

The default address added to the routing table is loopback set via csp_init(), and selecting the uart via the command line also
seems to add the same address to the routing table. So, the example adds two addresses to the routing table, both routes the same.

in csp_init():

csp_route_set(csp_conf.address, &csp_if_lo, CSP_NO_VIA_ADDRESS);

/* Also register loopback as default, until user redefines default route */

csp_route_set(CSP_DEFAULT_ROUTE, &csp_if_lo, CSP_NO_VIA_ADDRESS);

in csp_server_client.c

else if (default_iface) {
        csp_rtable_set(CSP_DEFAULT_ROUTE, 0, default_iface, CSP_NO_VIA_ADDRESS);

There appears no key-collision check to prevent adding two identical addresses with different intent.

The address selected for ping (also in the example) finds the first entry in the routing table, which is loopback.

Are there any non-trivial examples for using the routing table for setting destinations?

bob s.

johandc, 2021-06-02:
I see no issue here.

That routing table looks normal.

closed 2021-06-02 by johandc