libcsp/libcsp #179: memcpy before memory is allocated leads to segfault in csp_io.c csp_accept

schaefer01, 2021-01-06:
as before, I'm using the usart interface, connecting an embedded TI 432 c client to an ubuntu python server. 

the server side accept call for a received packet, in csp_io.c is csp_accept()

code snippet follows:
	csp_conn_t * conn;
	if (csp_queue_dequeue(sock->socket, &conn, timeout) == CSP_QUEUE_OK)
		return conn;

the queue dequeue does a memcpy from the socket to &conn, which points to nothing as memory hasn't been allocated,
the return stack is corrupted, and the return from csp_queue_dequeue results in a segfault

 csp_queue_dequeue vectors directly to int pthread_queue_dequeue(pthread_queue_t * queue, void * buf, uint32_t timeout)

code snippet follows:

/* Copy object to output buffer */
	   memcpy(buf, queue->buffer+(queue->out * queue->item_size), queue->item_size);

the API maps "buf" to "conn" and as previously noted, buf is an undefined pointer.

Your help as to what I should do to fix the problem would be helpful, so I don't have to wait on an official fix.

thank you for quick attention on this,

bob s.

schaefer01, 2021-01-06:
I'm thinking that maybe in your examples python bindings example, you are missing a call to 
  csp_conn_init()  and it's absence works with zmq.  Is this the right call to add?

schaefer01, 2021-01-06:
Got it, csp_conn_init is called by csp_init but the connection object remains in local space, so it is inaccessible.

schaefer01, 2021-01-06:
found the global connection variable in csp_conn.c, 

at the top of the file: static csp_conn_t * arr_conn;

I'm thinking this variable should be used instead of the local csp_conn_t * conn, for the accept call.

schaefer01, 2021-01-06:
I think I figured it out.  Let me know if I have it wrong, please.

this:
  csp_conn_t * conn;
  if (csp_queue_dequeue(sock->socket, &conn, timeout) == CSP_QUEUE_OK)

should be replaced by something like:

  conn = csp_conn_get_array(&size);
  if (csp_queue_dequeue(sock->socket, conn->rx_queue[0], timeout) != CSP_QUEUE_OK);

schaefer01, 2021-01-06:
argh.
in csp_connect.c, I found the global connections variable, at the top of the file.

/* Connection pool */
static csp_conn_t * arr_conn;

but in csp_conn_init(), this variable  is accessed as if it was an array:

int csp_conn_init(void) {
...
for (int i = 0; i < csp_conf.conn_max; i++) {
		csp_conn_t * conn = &arr_conn[i];
		for (int prio = 0; prio < CSP_RX_QUEUES; prio++) {
			conn->rx_queue[prio] = csp_queue_create(csp_conf.conn_queue_length, sizeof(csp_packet_t *),"rxQ");

I think I just aged 10 years when I read this. 
I'm pretty sure the results when executed are random writes into memory and will be indeterminate. help?

johandc, 2021-06-02:
You do have it wrong.

The local conn * variable is set by the call to dequeue. That's why we pass the address of the pointer to the dequeue.

closed 2021-06-02 by johandc