libcsp/libcsp #180: connection index is 0, csp_conn_allocate saves open state at index 1, things go south from there

schaefer01, 2021-01-08:
Again, I am using the usart driver, between ubuntu and a 432 embedded c processor

Inside csp_conn.c, csp_conn_allocate call is this:
or (int j = 0; j < csp_conf.conn_max; j++) {
		i = (i + 1) % csp_conf.conn_max;
		conn = &arr_conn[i];

so, the first allocate is at index 1, not index 0 as a rational human being would expect. 
      Index 0 is logically not possible.

now, if you followed this so far,

in python_bindings_example_server.py,  the call below returns the connection at index 0 

 conn = libcsp.accept(sock, libcsp.CSP_MAX_TIMEOUT) 

so, when a packet comes in and is enqueued at index 1, and then comes time to read to dequeue, 
the connection OPEN for the connection at index 1, but the read returns the connection for index 0.

the call to read in the python code, further down:

 packet = libcsp.read(conn, 100)

csp_read sees that the connection at index 0 is closed and the packet is not read because the connection is at index 1.

Does this problem description make sense?

Any help?

schaefer01, 2021-01-08:
I see what you did at the top of csp_conn_allocate with this variable:
  int i = csp_conn_last_given = 0;

 this fails on the first connection, perhaps it should be initiallized to -1;

schaefer01, 2021-01-08:
If you do that, i.e. init to -1, be sure to change the data type from unsigned to signed, else -1 won't work.
 
static uint8_t csp_conn_last_given should be int (why did you choose a char and not a full int?)

schaefer01, 2021-01-08:
When we launch the spacecraft (and its a success), I'd like to acknowledge you guys/gals. How about some help?

schaefer01, 2021-01-15:
possibly, index 0 is reserved for CONN_SERVER type connection and the other indices are for CONN_CLIENT type?
What is the purpose for CONN_SERVER?

johandc, 2021-06-02:
If you notice the Modulo operator, you see that conn index zero will also be used. It just starts at the last given + 1 which is the most likely candidate.
So this is not a bug.

closed 2021-06-02 by johandc