libcsp/libcsp #182: csp_kiss_rx in csp_if_kiss.c writes to bad pointer but even if it was a good pointer, memory wasn't allocated

schaefer01, 2021-01-13:
/* Valid data char */
((char *) &ifdata->rx_packet->id.ext)[ifdata->rx_length++] = inputbyte;

1. id.ext is an id, not a pointer, you might have meant rx_packet->data
2. rx_packet->data was not allocated memory to point to

the consequences of using the id as a pointer results in:
 *** stack smashing detected ***: <unknown> terminated

schaefer01, 2021-01-14:
ok, I found the line where the buffer was allocated, but the placement of inputbyte is still wrong.

schaefer01, 2021-01-14:
while I'm here, the header bits, id.ext aren't set, and this call in csp_kiss_rx() reads uninitialized bits:

ifdata->rx_packet->id.ext = csp_ntoh32(ifdata->rx_packet->id.ext);

Id.ext is set by the first 4 bytes after FEND, which should be read into rx_packet->data and

then read from rx_packet->data and only then manipulated by big/little endianness and placed into id.ext.

schaefer01, 2021-02-24:
I see what you did, id.ext is the value before the start of data and you index through the structure from one field (id.ext) to the next (data).
So, the pointer is good but the design not commented to explain what  you are doing.
The other part, not allocating data to be sized for a data packet,  I still believe to be true.

johandc, 2021-02-26:
Correct. We use the write through to fill out the packet header automatically in a zero copy maner.

What do you mean by:
not allocating data to be sized for a data packet,?

schaefer01, 2021-03-01:
has memory been allocated at the write-through, at packet->data[1] and beyond?

johandc, 2021-03-01:
Sure it has. All buffers are fully pre allocated at csp_buffer_init()

schaefer01, 2021-03-01:
you are right, I must have missed it..