libcsp/libcsp #183: csp_conn.c, csp_connect, assigns CSP_FCRC32 to local variable only, is not sent in packet flags

schaefer01, 2021-01-14:
declared in csp_connect(), these variables are not assigned to the connection:
csp_id_t incoming_id, outgoing_id;

		outgoing_id.flags |= CSP_FCRC32;
		incoming_id.flags |= CSP_FCRC32;

I verified this by generating a packet, transmitting the packet, and looking at the flags.

Note also, csp_if_kiss.c, csp_kiss_rx, csp_crc32_verify() is called always, without checking the flags for CSP_FCRC32

I'm still trying to verify if the outgoing crc matches the recalculation of the crc and have found at least one bug,
  csp_kiss_rx() passes the packet minus the crc bytes in the packet count (4 bytes less) but the csp_crc32_verify() function subtracts the 4 bytes AGAIN, so it is generating a checksum on a shorter packet than it should.

schaefer01, 2021-01-14:
I am also having issues with the function csp_ntoh32() in csp_endian(), which on ubuntu's gcc, does not reorder the bytes so much as generates a random number, but not for all uses in all places, which worries me.
I'm thinking of taking it out of inline and not having such a complex transform in a single code statement.

johandc, 2021-06-02:
This is due to double crc. Both on Mac layer and on network layer.

closed 2021-06-02 by johandc