libcsp/libcsp #185: shared memory doesn't work across threads, when using uart.

schaefer01, 2021-02-03:
A virtual address written to the fifoQ or pthread queue by one thread doesn't match the virtual-to-physical mapping when read in a different thread, and memcpy used to copy pointers across threads locks up the thread without, in my case, a segmentation fault.

 A C virtual memory pointer does not point to the same physical address from one thread to the next , so, a malloc’d buffer’s pointer saved into a  queue by one thread doesn’t match the same physical address when retrieved from the queue by a different thread.

For C calls needed to share memory across threads, see:
See: https://www.tutorialspoint.com/inter_process_communication/inter_process_communication_shared_memory.htm

schaefer01, 2021-02-03:
The uart doesn't really factor into the problem, its just not the same code path as when using zmq.

johandc, 2021-02-04:
That's wrong.

All threads share exactly the same virtual memory space. What you are referring to is inter process communication, which is spmetj completely different.

Your segfault must be caused by something else.

schaefer01, 2021-02-04:
Maybe, maybe not. This might be solely a Ubuntu problem.

I dumped queue address pointers in the main code, and again in csp_task_router and the pointers are the same address value but they don’t behave the same.

I’m creating additional tests and will get back to you.

schaefer01, 2021-02-05:
you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

schaefer01, 2021-02-05:
correction, the item size is 8 bytes and 16 bytes, the entry count is 20 and 25

On Feb 5, 2021, at 12:11 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

schaefer01, 2021-02-05:
nevermind for the buffer queue, I found the buffer pool size as a separate allocation, but I’m likely still right for the fifo queue size

On Feb 5, 2021, at 12:14 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

correction, the item size is 8 bytes and 16 bytes, the entry count is 20 and 25

On Feb 5, 2021, at 12:11 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

schaefer01, 2021-02-05:
the work done to properly size the objects in csp_buffer_init()

i.e.

// Ensure the csp_packet is correctly aligned (as it is not packed)
CSP_STATIC_ASSERT(CSP_HEADER_LENGTH == sizeof(csp_id_t), csp_header_length);
CSP_STATIC_ASSERT(sizeof(csp_packet_t) == 16, csp_packet);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, length) == 10, length_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, id) == 12, csp_id_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, data) == 16, data_field_misaligned);

should also be done for csp_qfifo_init()

On Feb 5, 2021, at 1:21 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

nevermind for the buffer queue, I found the buffer pool size as a separate allocation, but I’m likely still right for the fifo queue size

On Feb 5, 2021, at 12:14 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

correction, the item size is 8 bytes and 16 bytes, the entry count is 20 and 25

On Feb 5, 2021, at 12:11 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

schaefer01, 2021-02-05:
nope, this is not true, now that I looked at qfifo’s internals.
you are saving 2 pointers here, not the packet.

On Feb 5, 2021, at 1:28 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

the work done to properly size the objects in csp_buffer_init()

i.e.

// Ensure the csp_packet is correctly aligned (as it is not packed)
CSP_STATIC_ASSERT(CSP_HEADER_LENGTH == sizeof(csp_id_t), csp_header_length);
CSP_STATIC_ASSERT(sizeof(csp_packet_t) == 16, csp_packet);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, length) == 10, length_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, id) == 12, csp_id_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, data) == 16, data_field_misaligned);

should also be done for csp_qfifo_init()

On Feb 5, 2021, at 1:21 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

nevermind for the buffer queue, I found the buffer pool size as a separate allocation, but I’m likely still right for the fifo queue size

On Feb 5, 2021, at 12:14 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

correction, the item size is 8 bytes and 16 bytes, the entry count is 20 and 25

On Feb 5, 2021, at 12:11 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

schaefer01, 2021-02-05:
so, inside csp_qfifo_write(), in csp_qfifo.c

you declare a local variable, csp_qfifo_t queue_element;

then assign 2 malloc’d addresses of the interface and the packet data

queue_element.iface = iface;
queue_element.packet = packet;

you then push the local address of queue_element into a queue.

result = csp_queue_enqueue(qfifo[fifo], &queue_element, 0);

When it comes time to read the queue, the local address that points to the data is out of scope.

On Feb 5, 2021, at 2:58 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

nope, this is not true, now that I looked at qfifo’s internals.
you are saving 2 pointers here, not the packet.

On Feb 5, 2021, at 1:28 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

the work done to properly size the objects in csp_buffer_init()

i.e.

// Ensure the csp_packet is correctly aligned (as it is not packed)
CSP_STATIC_ASSERT(CSP_HEADER_LENGTH == sizeof(csp_id_t), csp_header_length);
CSP_STATIC_ASSERT(sizeof(csp_packet_t) == 16, csp_packet);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, length) == 10, length_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, id) == 12, csp_id_field_misaligned);
CSP_STATIC_ASSERT(offsetof(csp_packet_t, data) == 16, data_field_misaligned);

should also be done for csp_qfifo_init()

On Feb 5, 2021, at 1:21 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

nevermind for the buffer queue, I found the buffer pool size as a separate allocation, but I’m likely still right for the fifo queue size

On Feb 5, 2021, at 12:14 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

correction, the item size is 8 bytes and 16 bytes, the entry count is 20 and 25

On Feb 5, 2021, at 12:11 PM, Robert P. Schaefer <rps@mit.edu<mailto:rps@mit.edu>> wrote:

you are correct.

Here's the problem, the packet is larger than the buffers that hold the packet.

I instrumented the code and named each queue, you have buffer queues and fifo queues,
  each are sized to the size of the control information in the structure but not sized to hold the data, so the buffers are all undersized with
  respect to the packet data

the calling trace:

csp_qfifo.c -> csp_queue_create -> pthread_queue_create

csp_buffer.c -> csp_buffer_init -> csp_queue_create -> pthread_queue_create

pthread_queue_create, Qname=fifoQ, length=25, item_size=16, bytes=400           -> each queue memory chunk for the fifo is 25 bytes

pthread_queue_create, Qname=cspbufQ, length=20, item_size=8, bytes=160       -> each queue memory chunk for the buffer is 20 bytes

in csp_if_kiss.c, where I previously added an issue indicating that you weren’t saving the data to rx_packet->data,

  a PING packet (I cut the size down from your example to more easily trace in a print statement) is approximately 30 to 40 bytes.

so both on the buffer get, and th qfifo write, the allocated space is undersized.

ifdata->rx_packet = pxTaskWoken ? csp_buffer_get_isr(0) : csp_buffer_get(0);

...

ifdata->rx_packet->data[ifdata->rx_length++] = inputbyte;   -> this will overwrite the buffer because the packet is larger than the allocated memory to hold it.

...

csp_qfifo_write(ifdata->rx_packet, iface, pxTaskWoken);

johandc, 2021-02-06:
Why would interface and packet be out of scope? Did you allocate your interface or the packet on the stack?

schaefer01, 2021-02-22:
1. in reply to: "All threads share exactly the same virtual memory space."
   the C thread is called from a python thread, so the memory space is no longer shared. When shared memory is added the issue goes away.
2. in reply to: "Why would interface and packet be out of scope? Did you allocate your interface or the packet on the stack?" The original programmer allocated the packet on the stack.

johandc, 2021-06-02:
You cannot use the stack to allocate interface or packets. They need to have persistent scope. And All packets must use the built-in buffer pool.

closed 2021-06-02 by johandc