libcsp/libcsp #192: License compatibility with MIT or Apache 2.0

yashi, 2021-03-02:
Hello,

libcsp seems to be under LGPL 2.1.  FreeRTOS, OTOH, is under MIT and doesn't support dynamic linking.  IANAL, but this seems incompatible to me.  Does libcsp offer exception for static linking?

johandc, 2021-03-04:
Sure we can add that. To be frank I don't care what you use it for. You seem to be into this, maybe you can come up with a suggestion to a better licence. We just want the most open license that still maintains the original licence text.

yashi, 2021-03-07:
Thanks! I'll prepare an PR, then.

yashi, 2021-03-09:
hmmm... I've added a PR with a license exception but I'm not sure that's really what we want.  My intention was to allow linking against any free and open-source software.  But the exception I came up with allows any proprietary program to use libcsp without providing object files to users, which LGPLv2 requires.  It certainly allows user to do anything with libcsp, though.
comments welcome, as always.

closed 2021-03-09 by johandc

benner, 2024-06-20:
> Sure we can add that. To be frank I don't care what you use it for. You seem to be into this, maybe you can come up with a suggestion to a better licence. We just want the most open license that still maintains the original licence text.

I have similar question. I need to link against not open project to. v2.0 has MIT license but v1.6 branch, which is till most popular version, has LGPL.   Can I assume 1.6 is MIT too?

PR #193 (2021-03-09): README: Add license exception
The plain LGPLv2.1 distinguish two type of works
 - a work based on the library, and
 - a work that uses the library.

Usually, static linking is considered as former and dynamic linking is
considered as latter.  In addition to this, LGPLv2.1 section 6 states
that

  you may also combine or link a "work that uses the Library" with the
  Library to produce a work containing portions of the Library, and
  distribute that work under terms of your choice, provided that the
  terms permit modification of the work for the customer's own use and
  reverse engineering for [... cut]
commit ceafae3d (2021-07-26): README: Add license exception

The plain LGPLv2.1 distinguish two type of works
 - a work based on the library, and
 - a work that uses the library.

Usually, static linking is considered as former and dynamic linking is
considered as latter.  In addition to this, LGPLv2.1 section 6 states
that

  you may also combine or link a "work that uses the Library" with the
  Library to produce a work containing portions of the Library, and
  distribute that work under terms of your choice, provided that the
  terms permit modification of the work for the customer's own use and
  reverse engineering fo [... cut]