libcsp/libcsp #195: Example csp_server_client error

21212124, 2021-03-09:
Hi,

I am trying to run the examples but I have the following error:
The line executed:
` $ sudo ./waf configure build install --with-os=posix --enable-examples -v`

[image]

Why does this happen?
Thank you

closed 2021-03-15 by 21212124

JeremyDarcis, 2022-05-28:
Hello @21212124 !
I have the same error, can you tell me how you managed it please ? 
Thank you !

21212124, 2022-05-29:
Hi @JeremyDarcis,

I don't work on this project anymore but you can try these commands:

$ sudo apt install -y libzmq3-dev libsocketcan-dev
$ sudo ./waf configure --toolchain= --with-os=posix --prefix=install --enable-promisc --install-csp --enable-python3-bindings --enable-can-socketcan --with-rtable=static
$ sudo ./waf build install
$ sudo ./examples/buildall.py

Best regards.