libcsp/libcsp #206: "CSP_VERSION" Is not defined

arrooney, 2021-06-16:
Is this supposed to be "LIBCSP_VERSION" which is set in waf?

https://github.com/libcsp/libcsp/blob/0dbdc8e7a055462124cca10264d85dbeaf367210/include/csp/csp.h#L75

johandc, 2021-06-19:
No. The WAF build system is currently not maintained, since the switch to 2.0. we have people working on it.

As a workaround you can use mesok build system instead of simply modify WAF or the source to set CSP version to either 1 or 2.

yashi, 2021-07-25:
So, `CSP_VERSION` is a build option to tell the build system which version of library to build, and it's a single digit number.  `LIBCSP_VERSION` was the version number of the library, but it's not used and removed from the devel branch.

CSP_VERSION is not used at all, so the following patch will fix this.  But we still have more issues with `wscript`.
```diff
diff --git a/wscript b/wscript
index 52373a8..c8bb0ea 100644
--- a/wscript
+++ b/wscript
@@ -69,6 +69,7 @@ def options(ctx):
                   help='Set minimum compile time log level. Must be one of: ' + str(valid_loglevel))
     gr.add_option('--with-rtable', metavar='TABLE', default='static',
                   help='Set routing table type: \'static\' or \'cidr\'')
+    gr.add_option('--version2', action='store_true', help='Build CSP version 2')
 
 
 def configure(ctx):
@@ -121,6 +122,11 @@ def configure(ctx):
     ctx.define_cond('CSP_WINDOWS', ctx.options.with_os == 'windows')
     ctx.define_cond('CSP_MACOSX', ctx.options.with_os == 'macosx')
 
+    if ctx.options.version2:
+        ctx.define('CSP_VERSION', 2)
+    else:
+        ctx.define('CSP_VERSION', 1)
+
     # Add files
     ctx.env.append_unique('FILES_CSP', ['src/*.c',
                                         'src/external/**/*.c',
```
@johandc, do you want to merge this?  I can make a pr if you want to keep waf.  Or are you dropping waf support for v2.0?

johandc, 2021-07-25:
Gomspace would like to maintain WAF support for their build systems, and I don't see any reason the two build systems should not be able to coexist.

Surely I would merge that. Seems like a quick fix.

At space inventor I have moved to Meson.

johandc, 2021-07-26:
Thanks to yashi for his help

commit 24a9605d (2021-07-26): wscript: Add --version2 build option

We need `CSP_VERSION` defined in order to build and make the code
operate as either CSP version 1 or 2.  meson build already supports
this with the `version` project option with default to 1.

For waf, we add --version2 build option so that we can default to
version 1 without it, or specify it when building version 2.

This fixes #206.
closed 2021-07-26 by johandc