libcsp/libcsp #208: Dropped can frames can cause out of buffers problem

flaminggoat, 2021-07-07:
If csp_init is called with a configuration of 5 buffers libcsp can enter a state where it is unable to send any messages because all of the buffers are occupied by partially complete packet buffer elements. This occurs when several can frames are dropped, and then no more frames are received. When another can frame is received it will check the last access time of the PBUF elements and free them up, but until that happens all of the buffers are occupied.

I suggest that csp_init checks the number of buffers assigned is greater than PBUF_ELEMENTS defined here:
https://github.com/libcsp/libcsp/blob/53d740adbdf6c269a33ab68c2f548b03590dad8f/src/interfaces/csp_if_can_pbuf.c#L28
such that 1 buffer will always be available for other uses.

johandc, 2021-07-26:
Good point. I think the pbuf system is slated for removal in the near future anyways.

The alternative way is to assemble frames directly on CSP buffers without the need for PBUF. The additional header information in the PBUDs can easily be maintained in the reserved frame field of the packet structure. 

Now the issue we are talking about is flooding of incoming frames prevents outgoing packets. I believe this is something that can be avoided simply by adding a usage field to CSP buffer get. It would then be csp_buffer's responsibility to reserve packet buffers for higher priority stuff. Like for example outgoing packets.

johandc, 2021-12-05:
Fixed in spaceinventor/master

closed 2021-12-05 by johandc