libcsp/libcsp #209: Unify return value and errors

yashi, 2021-07-24:
As #23 suggested, we should use unified scheme for error numbers and return values.  We have `CSP_ERROR_NONE` defined as `0` in `csp_error.h` but `csp_queue`, `csp_semaphore`, and `csp_mutex` uses `1`, or `pdTRUE` / `pdPASS` as OK.

Because this is API breaking changes, it should only be for 2.0, as #23 suggested.

I'm creating this new issue becuase #23 is already closed.

johandc, 2021-07-25:
This will have to wait until a 2.1 or subsequent release. Since 2.0 is already in release candidate.

Theres a lot of cleanup that I would like to perform if I had the change to do over. 😊

yashi, 2021-07-25:
I didn't know 2.0 is that close.  There is no RC tag yet, right?

Would you give v2.1, or v3.0, tag to this?  Or did you mean we won't fix this?  I'm sure it's a huge change.

johandc, 2021-07-26:
I think we want to go with zero as OK and negative values as error.

This means the change needs to be in csp/arch which is used much more seldom than csp io. But yes it IS somewhat of a bigger change.

I don't want to ignore it, because it IS an inconvenient inconsistency. So I would like to address some of these internal stuff in a major release. 

The next API break will be in 2.1

johandc, 2021-10-30:
I fixed the csp_send return value issue now

johandc, 2021-11-16:
csp_mutex is gone now too

johandc, 2021-12-05:
The CSP_QUEUE_OK changed was non breaking, since only the macros was used it was only a matter of changing the value, and fixing the return of the FreeRTOS queue functions.

closed 2021-12-05 by johandc