libcsp/libcsp #210: Unable to build static routing table

yashi, 2021-07-25:
```
$ meson setup builddir
$ cd builddir
$ meson configure -Dwith_rtable=static
$ meson compile
  :
[44/45] Compiling C object libcsp.a.p/src_rtable_csp_rtable_static.c.o
FAILED: libcsp.a.p/src_rtable_csp_rtable_static.c.o 
ccache cc -Ilibcsp.a.p -I. -I.. -I../include -I/usr/include/pgm-5.3 -I/usr/include/libxml2 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -fPIC -isystem /usr/include/mit-krb5 -pthread -MD -MQ libcsp.a.p/src_rtable_csp_rtable_static.c.o -MF libcsp.a.p/src_rtable_csp_rtable_static.c.o.d -o libcsp.a.p/src_rtable_csp_rtable_static.c.o -c ../src/rtable/csp_rtable_static.c
../src/rtable/csp_rtable_static.c:26:27: error: ‘CSP_DEFAULT_ROUTE’ undeclared here (not in a function)
   26 | static csp_route_t rtable[CSP_DEFAULT_ROUTE + 1] = {};
      |                           ^~~~~~~~~~~~~~~~~
../src/rtable/csp_rtable_static.c:28:21: error: conflicting types for ‘csp_rtable_find_route’
   28 | const csp_route_t * csp_rtable_find_route(uint8_t dest_address) {
      |                     ^~~~~~~~~~~~~~~~~~~~~
In file included from ../src/rtable/csp_rtable_internal.h:21,
                 from ../src/rtable/csp_rtable_static.c:21:
../include/csp/csp_rtable.h:69:21: note: previous declaration of ‘csp_rtable_find_route’ was here
   69 | const csp_route_t * csp_rtable_find_route(uint16_t dest_address);
      |                     ^~~~~~~~~~~~~~~~~~~~~
../src/rtable/csp_rtable_static.c:40:5: error: conflicting types for ‘csp_rtable_set_internal’
   40 | int csp_rtable_set_internal(uint8_t address, uint8_t netmask, csp_iface_t *ifc, uint8_t via) {
      |     ^~~~~~~~~~~~~~~~~~~~~~~
In file included from ../src/rtable/csp_rtable_static.c:21:
../src/rtable/csp_rtable_internal.h:24:5: note: previous declaration of ‘csp_rtable_set_internal’ was here
   24 | int csp_rtable_set_internal(uint16_t address, uint16_t netmask, csp_iface_t *ifc, uint16_t via);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
../src/rtable/csp_rtable_static.c: In function ‘csp_rtable_set_internal’:
../src/rtable/csp_rtable_static.c:43:36: error: ‘CSP_ID_HOST_SIZE’ undeclared (first use in this function)
   43 |  if ((netmask != 0) && (netmask != CSP_ID_HOST_SIZE)) {
      |                                    ^~~~~~~~~~~~~~~~
../src/rtable/csp_rtable_static.c:43:36: note: each undeclared identifier is reported only once for each function it appears in
../src/rtable/csp_rtable_static.c:49:28: warning: variable ‘ri’ set but not used [-Wunused-but-set-variable]
   49 |         const unsigned int ri = (netmask == 0) ? CSP_DEFAULT_ROUTE : address;
      |                            ^~
../src/rtable/csp_rtable_static.c: In function ‘csp_rtable_iterate’:
../src/rtable/csp_rtable_static.c:65:21: error: ‘CSP_ID_HOST_SIZE’ undeclared (first use in this function)
   65 |    if (iter(ctx, i, CSP_ID_HOST_SIZE, &rtable[i]) == false) {
      |                     ^~~~~~~~~~~~~~~~
At top level:
../src/rtable/csp_rtable_static.c:26:20: warning: ‘rtable’ defined but not used [-Wunused-variable]
   26 | static csp_route_t rtable[CSP_DEFAULT_ROUTE + 1] = {};
      |                    ^~~~~~
ninja: build stopped: subcommand failed.
```

- `CSP_DEFAULT_ROUTE` was removed by 2684ecc42cc3f77015
- `CSP_ID_HOST_SIZE` was removed by f84704641a75c109b

johandc, 2021-07-25:
We should just remove static routing table and allocate static memory for the CIDR based table.

yashi, 2021-07-25:
I'm not sure how to allocate static memory for CIDR just yet.  Can I just remove the file for now?

commit fda08aec (2021-07-26): wscript: Make CIDR rtable default

The static rtable is not build-able right now in the `develop` branch.
Make the CIDR rtable default.  This is to sync with meson.build.

This fixes #210.
closed 2021-07-26 by johandc