libcsp/libcsp #212: Make Windows CI machine green

yashi, 2021-07-26:
With #211, I made all Mac OS X machines pass and green but was unable to make Windows CI machines work.  Hope someone with Windows experience fix it.

yashi, 2021-07-27:
@johandc, how about disabling the Windows CI machine on develop branch for now?  Because the CI is red all the time, we don't even bother to check it.  We can re-enable the machine once someone fixes it.

johandc, 2021-07-27:
Yes. The CI system was setup by a now ex GS employee, so not sure if I can do that. Is it possible to do by changing the file on GitHub alone, then go ahead and remove it. I will merge right away.

johandc, 2021-09-19:
This worked :)

I have access to Travis now, but we are out of credits. So i have talked to them about funding it as an OSS project, and hope to get reopened again.

closed 2021-09-19 by johandc

yashi, 2021-09-19:
Maybe it's time to move on to Github Action or some other CI services with free tier, buildkite, circleci, codeship...

johandc, 2021-09-20:
Do you have experience with any of said systems?

yashi, 2021-09-20:
Nop. But I have to move my free software repos to Github Action, anyway.  Do you have any preferences? Let's discuss it in #230.