libcsp/libcsp #220: memset() in csp_conn_free_resources()

yashi, 2021-07-29:
This is more of question rather than bug report.

In `csp_conn_free_resources()`, we `memset()` both `conn_lock` and `sport_lock` without calling `csp_bin_sem_remove()`.  What is the rational behind this?

johandc, 2021-07-31:
Good question...

It was added by per wulf here 
https://github.com/libcsp/libcsp/commit/11906eab97e6b4ab92409b41bd85de9286ef4da6

Seems like a bug to me.

I had an idea to use the queue system for buffer pools instead of arrays with a linear search. This would take it from O(n) to O(1) and also take care of locking. 

I believe this is already how the buffer pool is working.

Also the sport lock could be replaced by compiler based atomics.

johandc, 2021-09-19:
csp_conn_free_ressources have been removed now that we don't use any dynamic memory. Problem Fixed :)

closed 2021-09-19 by johandc