libcsp/libcsp #221: Split debugging and logging features

yashi, 2021-08-01:
It seems to me that the current code have mixed debugging feature and logging feature.  i.e) In order to print error log with `csp_log_error()`, we must enable `CSP_DEBUG`.  The function to set log level is called `csp_debug_set_level()`, instead of `csp_log_set_level()`.  It might be a good idea to separate them.

Actually, it'd be just a bunch of renames to do `s/csp_debug_/csp_log_/ig`.  We only have `csp_assert()` for debugging.

johandc, 2021-08-01:
So if I understand you correctly, you want to have CSP_DEBUG just control the CSP_ASSERT?

And rename debug_set_level to log_set_level. 

Your point is valid.

It's been a while since this systems was made, and if I should make it again I would do the following:

1. Use native assert.h I don't know why this should have ever been it's own function. Maybe for portability. But i have yet to find one that didn't have assert.h

2. Delete the entire log system.

3. Introduce a set of variables corresponding to each of the 7 socalled log levels we have today. This set will be initialised like all the other

johandc, 2021-08-05:
Okay yes we agree on the categories. This is what I meant with separate variables. One for each category. Like packet, drivers router and so on.

However I believe each category should be free to define its own levels, or bitfields (sub category). Like the error bit could be set simultaneously with a TX or RX bit or whatever flag that particular debug like will provide. As long as any bit is matched, it will produce an output.

So we can encourage the last four bits to always remain the same we have today. Err, warn, info, debug. And have the other fields in each category be free to choose from.

The call itself can be macroed or as a function call. To allow for external logging systems. However most systems should implement a way to catch stdout anyways.

We are moving more towards error counters, and we should be better to use them. Maybe together with logging for convenience.

johandc, 2021-09-26:
It's actually a problem with assert.h, because it uses __FILE__ which is considered harmful for reproducible builds. Because it contains the build folder. So one example could be if one developer builds his firmware in a short path /home/johan/git/proj, and another one has a longer name like /home/myverylongusername/git/mylongprojectname if you take this differece and multiply that with the numer of asserts in the code. The software image now does not fit the device. Changing the memory footprint and so on...

yashi, 2021-09-27:
https://github.com/libcsp/libcsp/blob/5344d8d00b55ed8674f2bd32083fc1844b29acdc/include/csp/csp_debug.h#L81

One advantage of `csp_assert()` is that it has a hook, `csp_assert_fail_action()`.  I assume we don't have it with standard `assert()`.  But, a simple question, you don't want to use `assert()` in the FM / Release build, do you?

johandc, 2021-09-27:
That's a tricky question.

I want reproducible builds, and i want to fly what i test. So YES, i want to fly the single "debug build". I just want it to fail nicely. So ideally each assert should be replaced with runtime recoverable faults, instead of the current action which is to crash the system. Even worse would be to run a build without debugging, and then just continue, potentially making more harm than a crash.

Another point is that `csp_assert_fail_action()` is rather useless, because i think it's individual which action should be made when an error is encountered. Sometimes you need to free some resources and return a known value. The issue with assert() is that it leaves the system in an unusable state.

The only advantage to assert is that it can be used to gain higher performance, by running rigorous testing in advance, and then removing all the error checking in a production release. However, this is NOT the philosophy i want to take. I want the code to be capable of detecting errors, and handle it gracefully. Even though it costs some performance to make the checks into runtime checks.

One method that we commonly use, is to have error counters being incremented, to let the operator aware that there is a code problem. I think the majority of the flash taken by the CSP library, actually consists of extremely verbose error messages. Many which could be replaced with simple counters or flags, raising the attention of the developer. After which, he could enable verbose debugging to include the printfs to stdout.

yashi, 2021-09-27:
> I want the code to be capable of detecting errors, and handle it gracefully.

I totally agree.  We want that in our flight software.  And the standard assert, or the mentality around assert isn't aligned to what we want.  This has been causing impedance mismatch in many projects.  Thus, my question: We do _not_ want to use the standard `assert()` because it doesn't do what we want.

> One method that we commonly use, is to have error counters being incremented

Yes, I'm aware of that.
And that is not CSP's job, IMHO.  Because it's nothing to do with network protocol.

I'd rather want libcsp to pass logging and error handing, or any other network protocol unrelated issues for that matter, down to the underlying OS or encapsulating framework.  To do that, I want libcsp to wrap as little arch specific functions as it needs, call them via wrapper in libcsp code base.

But this is a totally opposite direction if you want libcsp to be _the_ cubesat framework.

I guess my question boils down to: "Where do you want libcsp to be?"  This is a question I wanted to ask you for quite a bit. It doesn't related to the assert issue but it'd be nice if you can give us some future direction you have.

BTW, in Zephyr, we have a plug-able logging framework; we can write a backend to do the error counting and reporting on our required request timing.

johandc, 2021-09-29:
Okay, we agree thus far. Just one comment regarding the error counters. I think I do want to include the storage for those in the library. with the option of exporting them for mapping into users parameter system of choice. I will add a contrib file containing parameter mappings for our libparam. If zephyr has anything similar that could go in there too.

PR #304 (2021-12-06): Space Inventor Master
Fixes #221, #242, #241

The CSP core never emits calls to printf, instead it uses error counters for typical errors. And a global errno with codes for rare errors. The packet and protocol logs have been converted into flags, which enable the printf to stdout of these two. lock debugging have been wiped, i didn't feel it was necessary or used ever.

The new error counters will be able to catch many more overflow, and underruns than the old string based error log system. While the errno should simply always be 0. If it's not something is not setup correctly or there is a bu [... cut]
closed 2021-11-28 by johandc