libcsp/libcsp #225: Consider increasing the max data type for the router FIFO, at least on posix systems

arrooney, 2021-08-05:
I've found that the data type for the router FIFO can be a bottleneck for a busy CPU with lots of incoming data - the router FIFO tends to be the first thing to be overrun in stress tests even when set to 255. It doesn't look like there is any reason this can't be larger than a 8-bit when using pthreads. Might consider a uint16 here.

https://github.com/libcsp/libcsp/blob/53d740adbdf6c269a33ab68c2f548b03590dad8f/include/csp/csp.h#L56

johandc, 2021-08-05:
Sure let's increase a bit to 16 bits. If you make a pull request please do so on develop branch.

johandc, 2021-09-19:
This configuration parameter has been replaced in the new 2.0 static memory setup.

closed 2021-09-19 by johandc