libcsp/libcsp #226: Specification of protocols and services

dlr-ct, 2021-08-24:
Is there a public specification of the protocols that make up CSP and its included services?

Also, which of the included services (ping, ps, and so on) are required for normal operation?

yashi, 2021-09-15:
Hmm.. There used to be a pdf on GOMspace site.  But [google search][1] doesn't find it.  Instead searching for a plain [GS-CSP-1.1.pdf][2] shows us some but I'm not sure it's a legal copy.

@johandc, do you know where we can find it?
Should we have the file in our repo if GOMspace is OK with it?

[1]:https://www.google.com/search?q=GS-CSP-1.1.pdf+site%3Agomspace.com
[2]:https://www.google.com/search?q=GS-CSP-1.1.pdf

johandc, 2021-09-19:
Hehe, that document is one i made 10 years ago. I don't think it's up to date by any means. We really just need to add some new docs here on github. Preferbly in a morden format like .rst or markdown.

johandc, 2021-09-19:
I added the 2.0 specification here:

it only describes the base header and the CAN bus. The specification for I2C is not ready for 2.0. KISS and UDP interfaces are supported, but not documented yet.

https://github.com/spaceinventor/libcsp/blob/2.0/doc/CSP%202.0%20Specification%20-%20Ark1.pdf

closed 2021-09-19 by johandc