libcsp/libcsp #233: Build libcsp with FreeRTOS on CI

yashi, 2021-09-24:
I think it's a good idea to at least build libcsp against FreeRTOS on CI.

@johandc, I'm not familiar with FreeRTOS but I'd like to, first, test it locally and deply it to CI.  If you have a simple build steps, would you mind to share it with me?  It doesn't have to be a comprehensive at all.

johandc, 2021-09-24:
I think there is some kind of posix port of freertos available, that could be used to run basic tests.

yashi, 2021-11-15:
csp_arch.c on FreeRTOS for POSIX is done.  I'd like to host it under libcsp if you are OK with it.  I'd wanted to cross build for ARM  so that we can test FreeRTOS for ARM on qemu.   The blinky worked on QEMU but I couldn't integrate picolibc with it.   We can migrate to Non-POSIX later I assume.

https://github.com/spacecubics/libcsp-freertos

johandc, 2021-11-15:
Very nice indeed. Do you think it can run automatically with GitHub actions too?

I will setup the repo and provide you with write access

yashi, 2021-11-15:
Yes, I think I can.

commit 3583013f (2021-11-16): github: Add FreeRTOS build test

This commit adds a build test against the latest FreeRTOS Kernel.  The
test is located at github.com/libcsp/libcsp-freertos.  It currently
compiles and runs libcsp/examples/csp_arch.c against FreeRTOS POSIX
port.

This closes #233.

Please refer to libcsp-freertos for more detail.

https://github.com/FreeRTOS/FreeRTOS-Kernel
https://github.com/libcsp/libcsp-freertos
closed 2021-11-16 by johandc