libcsp/libcsp #234: csp_mutex_create_static() and csp_bin_sem_create_static() should return int

yashi, 2021-09-25:
Many csp function return values indicating success for failure. New functions, `csp_mutex_create_static()` and `csp_bin_sem_create_static()` should also follow the semantic.  I know both function has an out parameter, which indicate success for failure, but that's only for FreeRTOS.  POSIX and Zephyr won't need the second argument but nice to return `int`.

If you agree, I can post an PR.

yashi, 2021-09-25:
I might have sent too early.  Thread creation also returns a handle.  Let me think about it a bit more.

johandc, 2021-09-26:
I don't agree.

The reason is this. For static allocation there is no way a an allocation can fail. So there is no reason to make any error checking, which is actually the primary reason for using static allocation in the first place.

Using void where possible is much nicer and makes for cleaner and smaller code.

johandc, 2021-09-26:
As an example I want to make CSP send a void function too. The concept that once CSP send has been called that packet is out of here is much nicer than having to check for the case where an outgoing buffer was full, on every single call to send. 

Full output buffers should be handled by proper flow control instead. There is also a bug with CSP send having an inconsistent return type. So both can be solved.

yashi, 2021-09-26:
I agree that we want them to be allocation free.  But there are other restriction in RTOS.  For example,  [RTEMS sem_init()](https://docs.rtems.org/branches/master/posix-users/semaphore.html#sem-init) returns status code and has the following cases:
- **EINVAL**: The value argument exceeds SEM_VALUE_MAX (this is not a case for CSP because of binary nature)
- **ENOSPC**: A resource required to initialize the semaphore has been exhausted The limit on semaphores (SEM_VALUE_MAX) has been reached
- **ENOSYS**: The function sem_init is not supported by this implementation (this is not a case for CSP because all arch supports it)
- **EPERM**: The process lacks appropriate privileges to initialize the semaphore

QNX: http://www.qnx.com/developers/docs/qnxcar2/index.jsp?topic=%2Fcom.qnx.doc.neutrino.sys_arch%2Ftopic%2Fkernel_Semaphores.html

In practice it doesn't really fail as documented, so yes I agree them can return void.  I'll reopen it when I encounter a real problem.  Thank you for your time.

johandc, 2021-09-26:
ENOSPC: It seems like a poor implementation in RTEMS that it's not possible to pass the static memory along to the sem_init call. 

Reducing the amount of error checking in CSP would be really great. Often times there is no good solution to an error or a resource exhaustion in a network stack other than report the error and drop the packet.

closed 2021-09-26 by johandc