libcsp/libcsp #237: Removal of csp_endian.h

yashi, 2021-09-27:
Replacing `csp_hton32()` and others causes some problems:

- `htobe32()` and others are nonstandard
- The header file for them are `<sys/endian.h>` or `<endian.h>` instead of `<sys/types.h>`
- Because it's nonstandard, the function names varies,  `betoh32()` vs `be32toh()`
- We have standard `htonl()` and others in `<arpa/inet.h>` but lacks 64 bit versions, `htonll()`
- Zephyr doesn't support `htobe32()`
- Zephyr has nonstandard `htonll()` though, which FreeRTOS doesn't support it

It's a bit of chaos to me.

How about introducing arch specific headers instead of csp_endian?

yashi, 2021-09-27:
Nice one!
I'll try to add `<endian.h>` in Zephyr or just use picolib, which was something I wanted to integrate for a while anyway.

BTW, POSIX 2011? Care to share a link?

johandc, 2021-09-27:
https://news.ycombinator.com/item?id=7173813

johandc, 2021-09-27:
I can only recommend picolibc. I have been with that project since the breakout from newlib into its own entity. Keith Packard is a world class developer.

closed 2021-09-27 by yashi

dgonzalez97, 2023-01-24:
Hi! Im building with FreeRTOS for a STM32. When fixing the paths on the waf to compile using waf, some declaration errors came about. The errors are related to the endian.h

 7/34] Compiling src/crypto/csp_hmac.c
../src/csp_id.c: In function 'csp_id1_prepend':
../src/csp_id.c:53:8: error: implicit declaration of function 'htobe32' [-Werror=implicit-function-declaration]
   53 |  id1 = htobe32(id1);
      |        ^~~~~~~
../src/csp_id.c: In function 'csp_id1_strip':
../src/csp_id.c:73:8: error: implicit declaration of function 'be32toh' [-Werror=implicit-function-declaration]
   73 |  id1 = be32toh(id1);
      |        ^~~~~~~
../src/csp_id.c: In function 'csp_id2_prepend':
../src/csp_id.c:135:8: error: implicit declaration of function 'htobe64' [-Werror=implicit-function-declaration]
  135 |  id2 = htobe64(id2 << 16);
      |        ^~~~~~~
../src/csp_id.c: In function 'csp_id2_strip':
../src/csp_id.c:158:8: error: implicit declaration of function 'be64toh' [-Werror=implicit-function-declaration]
  158 |  id2 = be64toh(id2) >> 16;
      |        ^~~~~~~
cc1: all warnings being treated as errors

The endian.h im using is the one that comes from the arm compiler downloaded directly from the development /board/stm32cubeide_1.11.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.linux64_1.0.100.202210260954/tools/arm-none-eabi/include/machine/endian.h

Any clue why that could be? Thanks!

yashi, 2023-01-26:
Please create a new issue if you have a problem.  And please provide us a minimal reproducible example.  Without it we can't see your problem.

BTW I just built it against FreeRTOS with meson without a warning.
https://github.com/libcsp/libcsp-freertos

dgonzalez97, 2023-01-26:
Thanks for the fast answer, I really appreciate it. 
After doing some digging in your Changelogs i realized that at some point you changed the csp_endian.h files for the system endian.h. 
 Since im building for a CortexM and not for unix, those files arent included in my path, and the endian.h file of stm is way different from the one in unix. 
The fix was to downgrade to the branch of csp-1.6, where csp_endian.h still exists. I will however if I have some free time check for some clean implementation of the csp-2.0 on the CortexM

johandc, 2023-02-02:
Hello.

Your STM32 toolchain should come with a set of headers that provide platform specific endian functions. For example this has been supported by GCC for many years __builtin_bswap() and friends. If STM32 is based on newlib, there are bswap functions in there. Otherwise i heavily suggest compiling in your C-library into your project (see picolibc). This now has posix standard bwsap functions.