libcsp/libcsp #244: Consider using develop as the default branch on Github

yashi, 2021-09-27:
We mainly work on `develop` branch.  It'd be nice if the default branch is the one we work on. 
It also has benefits:
- less click for us to create a PR from web UI
- the project page shows what's really going on
- standard branch naming scheme for other users

While we are at it, if you are interested in, please use inclusive language for branch names: `master` -> `main`.

In conclusion, I propose:

- Rename `develop` -> `main`
- Rename `master` -> `1.0`
- Set `main` as the default

WDYT?

Reference:
- https://github.com/github/renaming
- https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/changing-the-default-branch

ps. For those with great muscle memory: [`git symbolic-ref`](https://git-scm.com/docs/git-symbolic-ref) is your help

johandc, 2021-09-28:
I think development should take place outside the default branch. If only it was a user preferences instead of a global preference. Because different developers could be focused on different developments.

I've seen a few projects lately that have used main instead of master. Is this a trend, or where does it come from?

So the short answer is, we should just get done with 2.0 and release that so that users sees the progress. No need to involve people in not yet released code on the front page. 

Finally since 2.0 is a release that breaks with quite a few tbings, we should create a branch for the 1.6 to be still maintained for a while after this. I suggest naming that 1.6

Also one other thing is that we have other companies that are contributing. My old colleagues at gomspace and satlab, who have yet to sign off on the 2.0 developments. Although they don't follow on GitHub these days. 

But as soon as we get there, we can start making smaller improvements against the master branch instead. And leave larger features for development branches.

yashi, 2021-09-28:
> I think development should take place outside the default branch.

If you prefer that way, that's fine.  Thank you for your time :smiley:

> main instead of master.

Someone asked to remove offensive term, master, with something else in the email [Rename offensive terminology (master)](https://lore.kernel.org/git/CAOAHyQwyXC1Z3v7BZAC+Bq6JBaM7FvBenA-1fcqeDV==apdWDg@mail.gmail.com/).  People argued, even the term "masterpiece" was brought up.  But it turns out that the term ["master" was from BitKeeper](https://mail.gnome.org/archives/desktop-devel-list/2019-May/msg00066.html), and refer to master/slave.  Git, Linux, Github, and many other projects and companies decide to remove many offensive terms, including blacklist.
https://github.com/torvalds/linux/commit/49decddd39e5f6132ccd7d9fdc3d7c470b0061bb.

closed 2021-09-28 by yashi

johandc, 2021-09-28:
Wow that debate is really a can of worms.

Personally I don't care what the name is, and it could just as well be main as master. 

I think it's crazy to go about deleting words from our vocabulary, just because they had once been used in a negative context. If some people actually stood forwards and said they are actually offended, we can talk about it. Once a certain percentage of the world population thinks it is.

But because some thinks that someone else, could potentially be offended, is not a reason. Even descendants after slaves on the US, the ones who may have a negative association with the word master, I think 99% are fine with the concept of having a master copy or a master piece that others are copied from.

yashi, 2021-11-16:
I just noticed that when I mention an issue in a PR, it doesn't close the mentioned issue when the PR is merged.  This is because develop branch is not the default branch.  Just another reason to make development branch as the default :wink:

johandc, 2021-11-16:
Okay. There you go.