libcsp/libcsp #251: Location of csp_autoconfig.h

yashi, 2021-09-29:
We no longer install `csp_autoconfig.h` so it doesn't really matter in practice but where should we put generated `csp_autoconfig.h`? Currently we have two locations:

- `include/csp_autoconfig.h`: waf
- `csp_autoconfig.h`: meson, cmake (due to following meson)

To find `csp_autoconfig.h`, we currently specify the top directory (build top to be precise) as a include directory.  But if we move it under `include/`, we have one less include directory to worry about.

Another thing we need to consider is that `csp_autoconfig.h` is implicitly mandatory for applications linking against `libcsp.a`.  That is, for meson based project for example, `csp_dep` must have a include directory containing `csp_autoconfig.h`.

We could align with other header files; under `include/csp/` and do `#include "csp/csp_autoconfig.h"`.  Just like any other header file we have.

Thoughts?

yashi, 2021-09-29:
It seemed a simple job turned out be a can of warm.... Meson doesn't even allow me to place the file in sub directory.
I'll tackle this after zephyr merge. Please keep this issue open and assign to me.  

Meanwhile, I just fixed `waf` to put `csp_autoconfig.h` in the build directory to align it with meson and cmake.

Buzzzz, 2023-08-17:
I am packaging libcsp for cross-compilation as a local debian package and have found that changing all include directives for this header to `#include <csp/autoconfig.h>` solves all the ugliness of needing the csp directory as another include path element.
This used the waf build. If the other builds work fine, can it be committed?

yashi, 2023-08-18:
@Buzzzz, would you please test the PR?

yashi, 2023-08-18:
note that
- Waf: Should work.
- Meson: Explicitly disabing libcsp.a installation.  Easy to fix but I'd love to wait for a comment from @johandc.  The urrent situation is broken; it installs header files but not libcsp.a.
- CMake:  No installation is supported.

Buzzzz, 2023-08-23:
Yashi-San, I have successfully completed a waf build and install from the PR. Looks very good. Thanks!

yashi, 2023-08-23:
Since @johandc have once agreeded renaming it in the past, I'm merging this.

yashi, 2023-08-23:
If someone need a proper install target in either Meson or CMake, please open a new issue.  Thanks.

PR #440 (2023-08-23): include: Rename csp_autoconfig.h to csp/autoconfig.h
This commit moves csp_autoconfig.h to build/include/csp/autoconfig.h. This allows us to remove the top directory from the include path.

This commit changes all header files and source files including csp_autoconfig.h.  Because of this, all three build system must also be changed in one go.

This fixes #251.
closed 2023-08-23 by yashi