libcsp/libcsp #257: Run tests on GitHub Action

yashi, 2021-10-01:
We haven't activated tests on GitHub Actions, yet.  There is no reason
not to for POSIX arch, at least.

johandc, 2021-10-01:
It should be worth the effort to make it run on arm through qemu too.

closed 2021-10-01 by johandc

yashi, 2021-10-01:
Do you want to run libcsp tests on ARM Linux? Or do you want FreeRTOS / Zephyr on ARM?

johandc, 2021-10-02:
On second thought. I think it is maybe too much work right now for CSP to go to target devices. That should be left for dedicated CI tests for each board.

But maybe just checking if the code compiles on arm and picolibc would be nice.

yashi, 2021-10-02:
Nah, it doesn't take much time to compile and run tests.   I will deploy libcsp tests on Zephyr on ARM on QEMU.  I don't know it's good enough to test only `lo` device but it's a start.  Zephyr development image is a rather big and I'm still learning how best to deploy it on GH Actions.

It'd be a fun project to write a QEMU device to make libcsp tests on both FreeRTOS and Zephyr talking each other on two QEMU instances.  That'd be ultimate goal. :smile: 

One problem is that we don't have unified test frame work.  Zephyr has a test framework and it works pretty well.  I'm sure FreeRTOS has it.  But we don't want to write a same test twice for sure.

commit 5fa30336 (2021-10-01): github: Run tests

Run csp_arch and csp_server_client on GitHub Actions just like Travis
CI does.

This closes #257.