libcsp/libcsp #259: Remove csp_thread

yashi, 2021-10-02:
This came up in https://github.com/libcsp/libcsp/pull/247#issuecomment-929391729
It's easier to remove but we need to decide how we run threads uniformly ​on different platforms.

yashi, 2021-11-01:
> So here's the deal. We do not provide csp_thread.h, but instead require the user to call csp_init, and then csp_router_work() (non blocking), or csp_router_loop() (blocking mode, takes over the calling task)

We have following `CSP_DEFINE_TASK()`
- `examples/csp_arch.c`
- `examples/csp_server_client.c`
- `src/csp_route.c` :: `csp_task_router`
- `src/csp_bridge.c` :: `csp_bridge`
- `src/interfaces/csp_if_zmqhub.c` :: `csp_zmqhub_task`
- `src/interfaces/csp_if_udp.c` :: `csp_if_udp_rx_task`

If we provide `csp_router_work()` and `csp_router_loop()`, we should also provide

- `csp_bridge_work()` and `csp_bridge_loop()`
- `csp_if_udp_rx_work()` and `and csp_if_udp_rx_loop()`
- `csp_zmqhub_work()` and `csp_zmqhub_loop()`

For the sample code under `examples/`, do you think it's OK to move each code under its own directory? So that we can have arch dependent files for each arch?

For example,

```
├── csp_arch
│   ├── csp_arch.c
│   ├── freertos.c
│   ├── linux.c
│   ├── win.c
│   └── zephyr.c
├── csp_server_client
│   └── csp_server_client.c
└── zmqproxy
    └── zmqproxy.c
```

In this case, we might want to rename the main file to `main.c`.

johandc, 2021-11-01:
Yes, Either that or:

```
 csp_arch.c
 csp_arch_linux.c
 csp_arch_freertos.c
 zmqproxy_linux.c
```

But i'm a little in doubt about how the FreeRTOS and Zephyr examples would run. Normally they should be included in another project in order for the OS to run, and be linked for the target. I think we may consider using an additional test repository to house some example projects which could be linked to some common target boards, like the SamExplained series.

yashi, 2021-11-01:
That's something I wanted to discuss about.

I don't know about FreeRTOS, but for Zephyr, I can have either in-tree samples or out-of-tree samples.  That is, Zephyr is application centric, meaning that you have to have application `.c` file with `CMakeLists.txt` and build Zephyr with it.   As a matter of fact, you can't build Zephyr as a library and later link against your app.  There is no `libzephyr.a` like most other RTOSes.   Plus, for now I have `csp_arch.c` for Zephyr under contrib but that can be any where.  All I have to have is `CMakeLists.txt` and some other related files along with `.c` file. 

So, it's your call.  We can make in-tree samples special for POSIX and have out-of-tree sample repos for RTOSes. Or, keep simple samples in-tree and maintain as much as we can.

johandc, 2021-11-01:
Okay.

I think it should be enough for libcsp, to include a few samples than can be compiled into a FreeRTOS or Zephyr application. For CI we should setup our own target specific repositories and routinely run them on our board targets.

I think an important destinction we need to make between the work and the loop functions are that the work call is purely asyncronous and has to return when the work is finished. Whereas the loop functions, should use a queue to wait for incoming data.

We still need for tasks to be able to sleep. Also RDP uses binary semaphores for synchronization between the router task and the RDP transmission task. So we do have some OS specific stuff left.

This means for example even though we have the router_work() function, it's still not possible to run CSP with a no-os board.

johandc, 2021-11-12:
How close is this to finished?

yashi, 2021-11-12:
FreeRTOS is the last one to remove.

johandc, 2021-11-13:
Thank you very much for this.

PR #293 (2021-11-13): Remove csp_thread support part 4
This is the last part of csp_thread removal.  With this, we can close #259.

This PR removes `csp_freertos_thread_create()`, which is the last arch.   and in csp_thread.h there was only `csp_sleep_ms()`, so I've renamed it to `csp_sleep.h`.  With this rename, all references to "csp_thread" are gone.

I have freertos version of csp_server_client_freertos.c pending. I'll create a PR after this is merged.

- part 1: #284
- part 2: #285 
- part 3: #289 

close #259
closed 2021-11-13 by johandc