libcsp/libcsp #260: Document about CMake

yashi, 2021-10-02:
We've added CMake support but not yet documented.

@johandc, where should I document about the build systems? `README.rst` or under `doc/`?

yashi, 2021-10-02:
#231

johandc, 2021-10-03:
first of all. I want to move to .md. 
You should put it into doc with links from readme. So it works on GitHub.
The major items from the  changelog should be promoted to the front page too.

yashi, 2021-10-03:
Just tried to see how well rst to gfm conversion works with [pandoc](https://pandoc.org/).  `literalinclude` doesn't work with gfm and GitHub is not willing to support it.

johandc, 2021-10-03:
Okay I think we should just copy the text referred to, rather than using fancy features like that one.

commit c4639df5 (2021-10-04): doc: INSTALL: Add how to build with the CMake build system

Add a new section using cmake to build libcsp.

This closes #260.
closed 2021-10-31 by johandc